Ultimaker/Cura #14048: Unable to Slice

Superbrain8, 2022-12-14:
### Application Version

5.2.1

### Platform

Windows 10

### Printer

Artillery Sidewinder X1

### Reproduction steps

Attempted to Slice a File with minor Details
Attempted to Slice a 20x20x20 Cube with no Details

### Actual results

File with Details causing a Slicing Error
File with no Details Causing the slicing process to get stuck

### Expected results

Sliced Gcode

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

Tried to Move the Files around the Buildplate, as suggested by other Issues opened
Both Files fail to Slice

Installation is a Clean Windows 10 Installation

[cura.log](https://github.com/Ultimaker/Cura/files/10229328/cura.log)
Cube with no Details
[FFPCTG_06_cube.zip](https://github.com/Ultimaker/Cura/files/10229331/FFPCTG_06_cube.zip)
Part with Minor Details
[FFPCTG_06_Filament_Card_PCTG.zip](https://github.com/Ultimaker/Cura/files/10229341/FFPCTG_06_Filament_Card_PCTG.zip)

GregValiant, 2022-12-14:
Thanks for the report and the files.
Someone from the Cura team will look at this.  I didn't have any trouble slicing either file but there are errors noted in your log file.  I don't think this is a slicing problem but there might be problems with your installation.

Superbrain8, 2022-12-15:
im a bit curious what the issue would be, did a fresh windows installation a week ago.

Superbrain8, 2023-02-09:
Figured out the issue.
had a fdmprinter.def in my definitions folder. this caused  some issues.

closed 2023-02-09 by Superbrain8