Ultimaker/Cura #14050: Travel moves to filled area origin and back after every top fill line

Bastel-Bodo, 2022-12-14:
### Application Version

5.2.1

### Platform

Windows 10

### Printer

Custom FFF printer (Original Prusa MINI+)

### Reproduction steps

I tried to slice an object consisting of a large, thin ring at its base - see the attachments for the 3mf project.
Some noteworthy slicing settings:

- Nozzle diameter: 0.6 mm
- Layer height: 0.25 mm
- Top/Bottom pattern: ZigZag
- Extra skin wall count: 1
- Infill: Grid, 15%
- Combing: Not in skin
  - Max distance without retract: 20 mm
- Bridge settings: On
  - Bridge sparse infill max density: 24%

### Actual results

On the first layer of top fill starting on the infill in the ring, cura inserts a travel move back to the origin of the filled area after each fill line. This behavior can be observed in layer 17 of the attached project.

In this screenshot, you can see the travel moves back to the top-left corner after each fill line:
[image]

The printed object came out okay, but the unnecessary travel moves did take quite some time.

### Expected results

The filled area should be filled in one go, i.e. after finishing one line of the filled patch, the slicer should continue immediately with the next adjacent fill line, without travelling somewhere else first.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[LampFrame-3mf-gcode.zip](https://github.com/Ultimaker/Cura/files/10230580/LampFrame-3mf-gcode.zip)

GregValiant, 2022-12-14:
Thanks for the report.
This excessive travel was a problem in 5.1 but I thought it had been fixed.  It's the Monotonic setting that is the problem.  I've noticed it my self but not in 5.2.1.
The Cura team will take a look.

gestalth, 2023-09-15:
I'm getting the same thing on 5.4. Changing the monotonic settings had no effect for me. Changing the combing mode from "Not in Skin" to anything else fixed it though. I was seeing the same pattern as above where it goes back to the same point after each short segment in a little jagged fill pattern. The print did finish but it created some visible artifacts from essentially ironing over the lightning infill and picking up some extra material on the nozzle.

Combing mode: Not in Skin -> ~7.5hr print time
Combing mode: Anything else -> ~6.5hr print time

Other move settings selected:
Retraction Minimum Travel: 1.5mm
Max Comb Distance: 30mm
Retract Before Outer Wall: Box checked
Avoid Printed Parts: Box checked
Avoid Supports: Box checked
Travel Avoid Distance: 0.625mm
Z Hop: Box not checked