Ultimaker/Cura #14051: Stay in one vertical area up to gantry height before traveling to reduce stringing

FalcoGer, 2022-12-14:
### Is your feature request related to a problem?

When printing a lot of disconnected vertical shapes, there is a lot of travel move between those shapes, causing stringing and taking time traveling when the printer could be printing.

### Describe the solution you'd like

The printer could instead print each vertical section for several layers depending on the gantry and print head geometry before traveling once to the next feature. This would require knowledge about the physical dimensions about the printer head, but would improve print quality and print time by a lot.

### Describe alternatives you've considered

N/A

### Affected users and/or printers

Everybody with a standard Cartesian coordinate (XYZ) FDM printer.

### Additional information & file uploads

Red: Current system
Blue: Proposed system

[image]

GregValiant, 2022-12-15:
Thanks for the thought you put into this.  The idea has been brought up previously.
The way it currently works is that if a print splits and there are two islands that need to be printed then Cura will do two layers here and two layers over there.  Once you get above doing double layers then crashes can happen.  My cooling duct end comes out behind the nozzle and is about 1mm above the print.  So I could maybe get 4 layers in before the duct hit the model.
My print head is about 90mm wide and 56mm deep.  So it needs to be around 40mm from an adjacent "island" in order to print something.  Every printer is going to be at least a little bit different but most have crash points well below the gantry.

FalcoGer, 2023-01-03:
Just very few layers at a time would improve quality and speed by a lot, particularly in models with lots of gaps. Every time the nozzle travels it has potential for stringing. Reducing the time the nozzle spends outside the model will help with that. Every extra layer printed before moving on to another section is one less potential defect.

Obviously every printer is different. Different regions of the print head may be required. Gantry size would be the maximum possible, but if the split is only a few tens of millimeters then only whatever amount the nozzle sticks out the bottom could be reasonably done without bumping into things. for me that's around 5mm, which still would equate to at least 10 layers. This could be even further improved with an extended nozzle.

The only drawback, besides the potential for misconfiguration and the one time requirement to measure out the print head and enter the values, is that printing small sections might not have had enough time to cool before the next layer is printed on top.