Ultimaker/Cura #14053: Non-specific error on Slicing

loshlee, 2022-12-14:
### Application Version

5.2.1

### Platform

macOS 12.6.1

### Printer

Custom FlashForge Creator Pro

### Reproduction steps

Loaded the .stl, then ensured surfaces were contacting build plate, then pressed "Slice"

### Actual results

Non-specific error. Link to but report is included in error message.

### Expected results

Slicing without error.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

You have to help me more with uploading to the site. This is frustrating.

GregValiant, 2022-12-15:
Thanks for the report.
The files need to be zipped to post here and Github has a 25mb limit on the size of Zip folders.  You can post a text file without zipping but pretty much everything else needs to be zipped.  If the file is over 25mb then you need to post it on a remote site like Google Drive and post the downloard link here.  It cannot be a site that requires an account to download.
To investigate this the Cura team will need a project file "(File | Save Project") and the Cura.log file ("Help | Show Configuration Folder").

There is a bug in the Windows installations that can keep a model from slicing when it is at the midpoint of the build plate.  The workaround is to move and/or rotate the model.  I haven't seen a report of it happening on Macs though.

loshlee, 2022-12-15:
bugreport_stl.3mf.zip
cura.log.zip

loshlee, 2022-12-15:
Thanks. I dragged them into this comment.
[bugreport_stl.3mf.zip](https://github.com/Ultimaker/Cura/files/10233527/bugreport_stl.3mf.zip)
[cura.log.zip](https://github.com/Ultimaker/Cura/files/10233529/cura.log.zip)

loshlee, 2022-12-15:
At least one other slicer had no problem with the same .stl file. Cura Mac 5.2.1 doesn't seem to be having this trouble with any other .stl files I've sliced with it.

GregValiant, 2022-12-15:
Thanks for the file.
I'm on Windows and It sliced for me on the first try.  After that I moved it around and rotated it in an attempt to get the slice to fail but I couldn't find an orientation where it would not slice.

The Cura team will check this out on a Mac system and see if they can duplicate the problem.

loshlee, 2022-12-15:
Thanks. I have successfully sliced many files that are similar in nature, this is the first that has failed.

closed 2024-03-01 by loshlee