Ultimaker/Cura #14070: moonracker

ward1975, 2022-12-17:
### Application Version

5.2.1

### Platform

window 10

### Printer

ender 3 s1 pro

### Reproduction steps

i was on mark place in cura 

### Actual results

hi there i had this problem before with moonracker its not uploading to my sonic pad i tried and tried but it still dose not work. I slist the xyz cube and i tried 4 times its coming up with the same error and come up with a report page 
[image]
 
[image]

### Expected results

can this be sorted out because i love klipper  

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]

GregValiant, 2022-12-17:
Thanks for the report.
If it is a slicing error the report comes here.  If it is a Moonraker error then the slice goes to the developers who created the plugin. If you go to the Marketplace and to the Moonraker icon there is a link to the emtrax people.

ward1975, 2022-12-17:
Ok thank you I do that sorry for your time

GregValiant, 2022-12-18:
Not a problem.  It's the nature of open source software that there can be many contributors and it isn't always clear who is responsible for a particular part.  

I'll go ahead and close this then.  If Emtrax comes back and says that it's a problem on the Cura end then this can be re-opened.

closed 2022-12-18 by GregValiant