Ultimaker/Cura #14071: Sliced model missing details depending on orientation

Ltek, 2022-12-17:
### Application Version

5.2.1

### Platform

Windows 10

### Printer

Sidewinder X1

### Reproduction steps

Screenshots show when in prepare vs sliced. When the model is horizontal on the bed, Cura misses critical model details. This is the SAME STL imported into Cura 5.2.1 and duplicated - changed orientation, then sliced. you can see it is missing the center debossed area (which is identical depth to the top debossed area), and the holes on the back that are connection points to the a separate STL which is the 2nd part of the model.
[image]
[image]
[image]

### Actual results

Areas of the model / STL are not being rendered when sliced

### Expected results

All details should be sliced in any/all orientations

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

see above

GregValiant, 2022-12-17:
Thanks for the report.
As I said over on the UM forum it may well be that "Remove All Holes" is enabled.
Use the "File | Save Project" command, zip the 3mf file, and post the zip file here.

Ltek, 2022-12-18:
attached the 3MF file I'm using
[ASX1_CloseHoleProblem.zip](https://github.com/Ultimaker/Cura/files/10252956/ASX1_CloseHoleProblem.zip)