Ultimaker/Cura #14073: wont slice with tree supports on 5.1.1

DasWarlock, 2022-12-18:
### Application Version

5.1.1

### Platform

windows 11

### Printer

anycubic kobra max

### Reproduction steps

tried to slice a plate with tree supports and each time it would fail slicing. restarted cura 4x and restarted computer 2x and it still wont slice.

### Actual results

sliceing fails

### Expected results

it should complete the slice

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log.zip](https://github.com/Ultimaker/Cura/files/10253041/cura.log.zip)
[CFFFP_DinnerPlateStraight_v03.zip](https://github.com/Ultimaker/Cura/files/10253042/CFFFP_DinnerPlateStraight_v03.zip)

Sonicboom247, 2022-12-18:
same issue on 5.2.1, slicing a large object with tree supports, 279mm surfboard and error slicing failed with an unexpected error. switching to normal slices fine. object was 279mm long, reduced to 220 to test and slice was fine, went back up to 250mm and error returned. not sure if it just can't place supports somewhere.  I reduced Tree support Branch angle from 40 to 30 and it sliced successfully.

MariMakes, 2023-05-12:
Hey @DasWarlock,

Welcome to the UltiMaker Cura Github 🚀 
Sorry, it took us a while to get back to you 😞 

What an interesting model to use a 3D printer for 🤔 
It's also a tricky 3D print in general, regardless of your support choice. 

I took a look and it crashes for me in the recent stable version of Cura. 

But... I also took a look in a nightly version and the crash seems to resolved. 
The upcoming Cura 5.4 contains the new tree supports of Thomas Rahm and this crash seems to have been resolved there. 
[image]

To me, it doesn't seem like a bug in Ultimaker Cura anymore, since we have a solution in place. 
Apologies for any inconvenience caused by our delayed response. 
I'll be closing this issue. I hope you understand.

closed 2023-05-12 by MariMakes