Ultimaker/Cura #14076: Internal GUI components have no sanity checks governing their size

The--Captain, 2022-12-18:
### Application Version

4.6.2

### Platform

Windows 10

### Printer

Custom

### Reproduction steps

Cause a problem with a profile that generates an error message with 30 or 40 lines of errors upon startup
or
Add more than seven or eight extruders

I'm sure there are other examples

When certain graphical elements within Cura (e.g. The startup error message diaglog, or the left-hand extruder selector) contain a large amount of data or elements, no constraints appear to govern them.  I have had to download custom screen stretch-and-scroll software in order to access the close box of the startup error dialog when it grew so large it was rendered outside the window borders of Cura.  This is a bug.  No elements of the software should render outside the window borders of the software.

[image]

### Actual results

Element that should appear onscreen render outside the window border, making them impossible to access without specialized 3rd party screen manipulation software.  This is especially annoying when the startup error dialog contains enough error lines to cause the close 'X' box of the error dialog to render outside the window border, making it impossible to close (without installing software that allows you to manipulate window sizes and scroll areas)

### Expected results

GUI elements should be constrained or scrollable if there is too much data to be displayed within the current window.

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

Are you trying to restrict legit bug reports by requiring users to attach files that were not generated and are not relevant?  Because it sure seems so.  Just letting you know why my log file is empty, in case you needed help figuring it out.

[cura.log](https://github.com/Ultimaker/Cura/files/10253636/cura.log)

MariMakes, 2022-12-20:
Hey @The--Captain,

Thanks for your report 👍
Can you share why you are using so many extruders?
Do you have that many extruders on your printer? 
Do you use it as a workaround for missing functionality?

We don't meet a lot of makers who have that many extruders so it's hard to find what's important for them. We see that in a single day we have less than 50 slices on 3D printers with 3 or more extruders, compared to 500,000+ slices with 3D printers with 1 or 2 extruders. 

I'll bring it up to the team to see what they can do to improve it. But I expect that if it's not a quick fix that it will not get a high priority to implement. Fingers crossed 🤞

The--Captain, 2022-12-31:
Yes, eight extruders, in a many-to-one configuration.  It's an MMU.  I like to be able to generate multicolor prints, also the convenience of rarely having to swap reels to print in a different color or material cannot be overstated ;-) 

https://www.thingiverse.com/thing:4589255

I use Klipper firmware, and color swaps are as simple as a couple of lines in the start and end gcode for each extruder

The--Captain, 2023-03-07:
Monitor scaling is a total workaround, and still will not fix the problem if the window output is large enough.  Also, you seem potentially confused, as I am not talking about a lack of responsiveness, but a failure to implement proper bounds checking in your window rendering functions.

This should not be closed until your team is confident that large output does not overflow the window in which it is rendered, and ALL output is enabled to be viewed.  I can almost guarantee your GUI libraries include a scroll function for this kind of thing, unless you wrote them yourselves (please tell me you did not do that)

Has this been fixed, or not?

fieldOfView, 2023-03-07:
Hey, I am an external contributor, I am not an UltiMaker employee, so please do not get made at UltiMaker because of things I say. 

I just want to bring to your attention that you come across as quite entitled. It is almost as if you think the Cura team owes you. Does it?

It is awesome if you *respectfully* bring issues to their attention, but there is entirely no need to be condescending or demand that your issue get fixed when the team tells you that they will not have the resources to prioritize your issue.

The--Captain, 2023-03-07:
They can fix it or not, but surely you must admit that the reply about "responsiveness" was not related to this issue.   This is basic GUI coding.  When I write code that needs to display potentially huge info than might not fit, I use the scroll function (or implement one myself on the rare occasion the library omits this function).  

The left sidebar is not the only place that this bug appears.  The dialog that displays startup errors (profile-related, etc) will also exhibit this bug when the information overflows the allocated space for rendering, so the Cura team is just ignoring bounds checking right and left.  It makes one shudder to think what other common sense programming techniques are being thrown out the window that are not as evident.

FYI, your tone policing is unwarranted.  Would you hesitate to complain if someone forgot to install tires on the city bus you happen to ride, just because it's publicly funded?  That's not "condescending", it's pointing out a ridiculous lack of common sense.

And I know who you are.  I love your plug-ins.  Call me crazy, but I'd think you would want them to be installed on as stable software as possible, so people don't associate the errors of the Cura team with your own coding skill.