Ultimaker/Cura #14077: Crashes constantly

luckykismat86, 2022-12-18:
### Application Version

5.2.1

### Platform

windows 10

### Printer

Anycubic Kobra Max

### Reproduction steps

step one Clicked on market place

### Actual results

nothing program crashes.

### Expected results

market place pops up to be used.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

.

MariMakes, 2022-12-20:
Hey @luckykismat86,

Welcome to the Ultimaker Cura Github 🚀 
Thanks for your report 👍

Sorry to hear that you are stuck 😖 

I have a couple of questions to narrow it down.
	1. Do you have multiple monitors? Does Cura start if you disable one of the monitors?
	2. Do you have GPU Tweak OSD overlay enabled? Does Cura start if you disable that software?
	3. Do you have a Ryzen Graphics card? Can you try this workaround? https://support.makerbot.com/s/article/1667412437627
	4. Do you have the Nahimic audio enhancement service enabled? Can you try this workaround? https://github.com/Ultimaker/Cura/issues/12102#issuecomment-1157907665

luckykismat86, 2022-12-20:
> this workaround? #12102 (comment)

1. No just a single monitor
2. No OSD
3. Yes but fix did not work.  I also tried forcing using graphics card and
CPU graphics.
4. No Nahimic