Ultimaker/Cura #14079: Slicing fails in 5.2.1, but slices fine in 4.4.1

jasonrohrer, 2022-12-18:
### Application Version

5.2.1

### Platform

Ubuntu 20.04.5 LTS

### Printer

Ultimaker S3

### Reproduction steps

1. Open Cura 5.2.1
2. Load the project.
3. Slice

### Actual results

Slice fails with Unexpected Slicing Error

### Expected results

This slice works fine in 4.4.1 (which is the repository version of Cura in Ubuntu 20.04.5 LTS).

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[UMS3_pretty3DTree_slicer_bug.zip](https://github.com/Ultimaker/Cura/files/10254327/UMS3_pretty3DTree_slicer_bug.zip)
[image]
[cura.log](https://github.com/Ultimaker/Cura/files/10254334/cura.log)

jasonrohrer, 2022-12-18:
Other notes:

If you scale it to 200% (instead of 300%), the slice completes successfully and quickly.

If you scale it to 400%, it takes FOREVER (like... 20+ minutes before I give up and kill it?)  It's too slow to be usable, especially for such a simple model.  I never waited for it to finish, so I'm not sure if it eventually reaches a slicing error or not.

In 4.4.1, all scales (200%, 300%, and 400%) complete successfully and quickly with PVA supports.

This model also slices fine in FlashForge's FlashPrint slicer.

MariMakes, 2022-12-20:
Hey @jasonrohrer,

Thanks for your report and the extra investigation👍
This looks like a super complicated model 👀 
I'm super curious what it's used for. 

[image]

This mesh seems faulty
Have you tried doing a quick mesh fix with the Meshtools Plugin? 
You can get it here: https://marketplace.ultimaker.com/app/cura/plugins/fieldofview/MeshTools

I also noticed that I am able to slice the model if I rotate it by 180 degrees and move it away from the center.
In your case it might also speed up the printing time if you move it close to the switching bay 😉 
[image]

This seems to be a duplicate of https://github.com/Ultimaker/Cura/issues/13643
We have a ticket on our backlog with the intent to improve this behavior.
It has to do with rounding errors, and that is probably why you are seeing different behavior with different sizes. 

I'll be closing this issue as a duplicate but you can follow the progress here: https://github.com/Ultimaker/Cura/issues/13643

jasonrohrer, 2022-12-21:
Thanks!

I knew that this might be a dupe... just wanted to report another example, just in case.

It's a sculpture that serves no purpose other than looking interesting.

And yes, there are "errors" in the mesh.... it's output from Structure Synth, and is composed of overlapping cubes.

However, slicers generally seem to be okay with errors like this.  Cura usually handles it just fine.

jellespijker, 2023-01-17:
@MariMakes this is unreleated to #13643 it crashes in an other area of the code.

[support.cpp#L1297](https://github.com/Ultimaker/CuraEngine/blob/701e72c0012544d60d35c653f35a9a623a289cc1/src/support.cpp#L1297)

I will reopen this ticket

Vandresc, 2023-01-30:
Slices perfectly for me in the latest main build. Will close for now, please re-open if the issue still persists for you.

closed 2023-01-30 by Vandresc