Ultimaker/Cura #14080: After slicing: nothing to show in preview

gabber130779, 2022-12-18:
### Application Version

5.2.1

### Platform

Windows 11

### Printer

BIQU Hurakan

### Reproduction steps

1. Slice the model
2. Klick on slice button
3. klick on Preview

### Actual results

Nothing show in Preview.
[image]

### Expected results

show sliced model in preview
[3DBenchy BasicFil PLA lw 0.4mm lh 0.2mm if 8% ext1 200C bed 60C.zip](https://github.com/Ultimaker/Cura/files/10254620/3DBenchy.BasicFil.PLA.lw.0.4mm.lh.0.2mm.if.8.ext1.200C.bed.60C.zip)
[cura.log](https://github.com/Ultimaker/Cura/files/10254626/cura.log)

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[image]

cassidyjames, 2023-02-12:
I'm also getting the same symptoms here in Cura 5.2.2 on Linux. I'm on 11th-gen Intel Xe graphics, if that helps, too.

MariMakes, 2023-02-13:
Hey @gabber130779,

We've added a ticket to the backlog with the intent to improve this. 
For internal reference CURA-10075

Thanks for the report! 👍 

@cassidyjames, we were under the impression it just happened with this specific printer type. Do you mind sharing your project file? We can add it as a test case to test if when there is a solution in place. 

Do you have a project file for us? It contains the printer and settings we need for troubleshooting. 
To save a project file go to File -> Save project. Please make sure to .zip your project file. For big files, you may need to use WeTransfer or similar file-sharing sites.

You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser.

cassidyjames, 2023-03-15:
Hey @MariMakes weirdly this only happens to me on one computer, but it happens with every print file. On my other machines with the same printer, it works fine. I can try to reset the app and see if it persists, and if so, get you a project file.