Ultimaker/Cura #14082: Initial Temperature Defaults RepRap Gcode

Jak7ob, 2022-12-19:
### Application Version

5.2.1

### Platform

Windows 10

### Printer

Prusa I3 MK2

### Reproduction steps

1. Upgraded Cura from 4.4.1 to 5.2.1
2. Uploaded all default parameters and settings according to older version
3. Copy over customer profiles for printer and filament
4. Updated Start Gcode and End Gcode settings according to personal script
5. Bed and nozzle temperatures are included in Gcode
6. Set all parameters for printer and add model to bed to slice
7. Slice and save model to drive
8. See there is a M190 S60  M104 S212  M109 S212 at the beginning
9. Try to trouble shoot issue but have no luck after 45 min.

### Actual results

There is a M190 S60  M104 S212  M109 S212 at the beginning

### Expected results

With 4.4.1, when I had "M104 S140 ; set extruder no-ooze temp" and "M140 S40  ; set bed PINDA warmup temp" included in my code, Cura did not try to set initial temperatures for the nozzle and bed. I would expect Cura to allow me to set my own temperatures as there is no reason to heat everything to print temperature for a calibration sequence on the printer.  
I would be happy to discuss this issue at length as even when I did include the exact thing Cura was adding at the beginning of the Gcode, it became redundant and added it again.
My current fix is to just manually edit the Gcode to remove this addition but this has become tedious and should not be necessary.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/10255042/cura.log)

Asterchades, 2022-12-19:
Are you able to show your entire Start GCode?

My understanding is that it used to be Cura would look for the presence of temperature commands (M104, M109, M140, and M190) to decide whether it would apply its own or not. Either this was wrong or it has since been changed to look instead for the initial layer placeholders instead. So the absence of `{material_print_temperature_layer_0}` will cause it to heat the nozzle itself while the absence of `{material_bed_temperature_layer_0}` will do the same for the bed. 

_It's also possible it's just looking for any temperature placeholder for the respective heaters, rather than just the initial layer ones._

Jak7ob, 2022-12-20:
I have three different variants that I tried when I was looking at fixing
this bug. My "Set1 (Default)" is how I would normally generate the start
Gcode and when I first saw the issue. "Set2" was my first attempt at
correcting this issue with the order of setting the temperature.
"Set3" is the most obvious generated code that there is an issue.
I do not know any additional settings or changes that I would be able to do
and fix the issue from my side. My best guess, there is some issue with the
condition that if nozzle and bed temp is set, do not set default temps at
start.

I hope this helps, let me know if you have any questions.

Jacob

Start Gcode
M104 S140 ; set extruder no-ooze temp
M140 S40  ; set bed PINDA warmup temp
G21 ; set units to millimeters
G90 ; use absolute positioning
M83 ; use relative distances for extrusion
M300 S40 P10 ; chirp
G28 W ; home all without mesh bed level
G80 ; mesh bed leveling
M104 S212 ; set extruder temp
M140 S60 ; set bed temp
M190 S60 ; wait for bed temp
M109 S212 ; wait for extruder temp
M300 S25 P10 ; chirp
G1 Z0.250 F7200.000 ; go outside print area
G1 X50.0 E80.0  F1000.0 ; intro line
G1 X160.0 E20.0  F1000.0 ; intro line
G1 Z0.200 F7200.000 ; intro line
G1 X220.0 E13 F1000.0 ; intro line
G1 X240.0 E0 F1000.0 ; intro line
G92 E0.0 ; reset extruder distance position
M82 ;absolute extrusion mode

Generated Gcode
T0
M190 S60
M104 S212
M109 S212
M82 ;absolute extrusion mode
M104 S140 ; set extruder no-ooze temp
M140 S40  ; set bed PINDA warmup temp
G21 ; set units to millimeters
G90 ; use absolute positioning
M83 ; use relative distances for extrusion
M300 S40 P10 ; chirp
G28 W ; home all without mesh bed level
G80 ; mesh bed leveling
M104 S212 ; set extruder temp
M140 S60 ; set bed temp
M190 S60 ; wait for bed temp
M109 S212 ; wait for extruder temp
M300 S25 P10 ; chirp
G1 Z0.250 F7200.000 ; go outside print area
G1 X50.0 E80.0  F1000.0 ; intro line
G1 X160.0 E20.0  F1000.0 ; intro line
G1 Z0.200 F7200.000 ; intro line
G1 X220.0 E13 F1000.0 ; intro line
G1 X240.0 E0 F1000.0 ; intro line
G92 E0.0 ; reset extruder distance position
M82 ;absolute extrusion mode
Start Gcode
G21 ; set units to millimeters
G90 ; use absolute positioning
M83 ; use relative distances for extrusion
M104 S140 ; set extruder no-ooze temp
M140 S40  ; set bed PINDA warmup temp
M300 S40 P10 ; chirp
G28 W ; home all without mesh bed level
G80 ; mesh bed leveling
M104 S212 ; set extruder temp
M140 S60 ; set bed temp
M190 S60 ; wait for bed temp
M109 S212 ; wait for extruder temp
M300 S25 P10 ; chirp
G1 Z0.250 F7200.000 ; go outside print area
G1 X50.0 E80.0  F1000.0 ; intro line
G1 X160.0 E20.0  F1000.0 ; intro line
G1 Z0.200 F7200.000 ; intro line
G1 X220.0 E13 F1000.0 ; intro line
G1 X240.0 E0 F1000.0 ; intro line
G92 E0.0 ; reset extruder distance position
M82 ;absolute extrusion mode

Generated Gcode
T0
M190 S60
M104 S212
M109 S212
M82 ;absolute extrusion mode
G21 ; set units to millimeters
G90 ; use absolute positioning
M83 ; use relative distances for extrusion
M104 S140 ; set extruder no-ooze temp
M140 S40  ; set bed PINDA warmup temp
M300 S40 P10 ; chirp
G28 W ; home all without mesh bed level
G80 ; mesh bed leveling
M104 S212 ; set extruder temp
M140 S60 ; set bed temp
M190 S60 ; wait for bed temp
M109 S212 ; wait for extruder temp
M300 S25 P10 ; chirp
G1 Z0.250 F7200.000 ; go outside print area
G1 X50.0 E80.0  F1000.0 ; intro line
G1 X160.0 E20.0  F1000.0 ; intro line
G1 Z0.200 F7200.000 ; intro line
G1 X220.0 E13 F1000.0 ; intro line
G1 X240.0 E0 F1000.0 ; intro line
G92 E0.0 ; reset extruder distance position
M82 ;absolute extrusion mode
Start Gcode
M190 S60
M104 S212
M109 S212
M104 S140 ; set extruder no-ooze temp
M140 S40  ; set bed PINDA warmup temp
G21 ; set units to millimeters
G90 ; use absolute positioning
M83 ; use relative distances for extrusion
M300 S40 P10 ; chirp
G28 W ; home all without mesh bed level
G80 ; mesh bed leveling
M104 S212 ; set extruder temp
M140 S60 ; set bed temp
M190 S60 ; wait for bed temp
M109 S212 ; wait for extruder temp
M300 S25 P10 ; chirp
G1 Z0.250 F7200.000 ; go outside print area
G1 X50.0 E80.0  F1000.0 ; intro line
G1 X160.0 E20.0  F1000.0 ; intro line
G1 Z0.200 F7200.000 ; intro line
G1 X220.0 E13 F1000.0 ; intro line
G1 X240.0 E0 F1000.0 ; intro line
G92 E0.0 ; reset extruder distance position
M82 ;absolute extrusion mode

Generated Gcode
T0
M190 S60
M104 S212
M109 S212
M82 ;absolute extrusion mode
M190 S60
M104 S212
M109 S212
M104 S140 ; set extruder no-ooze temp
M140 S40  ; set bed PINDA warmup temp
G21 ; set units to millimeters
G90 ; use absolute positioning
M83 ; use relative distances for extrusion
M300 S40 P10 ; chirp
G28 W ; home all without mesh bed level
G80 ; mesh bed leveling
M104 S212 ; set extruder temp
M140 S60 ; set bed temp
M190 S60 ; wait for bed temp
M109 S212 ; wait for extruder temp
M300 S25 P10 ; chirp
G1 Z0.250 F7200.000 ; go outside print area
G1 X50.0 E80.0  F1000.0 ; intro line
G1 X160.0 E20.0  F1000.0 ; intro line
G1 Z0.200 F7200.000 ; intro line
G1 X220.0 E13 F1000.0 ; intro line
G1 X240.0 E0 F1000.0 ; intro line
G92 E0.0 ; reset extruder distance position
M82 ;absolute extrusion mode

closed 2022-12-20 by Asterchades

Asterchades, 2022-12-20:
None of your Start GCodes include the placeholders that Cura is looking for, so it's adding them on its own before everything else as a precaution.

Change the `S60` on your bed heating lines to `S{material_bed_temperature_layer_0}` and the `S212` on your tool heating lines to `S{material_print_temperature_layer_0}`. If there's a specific reason you're hard-coding the temperatures to the start script instead of setting it in either the print or material profiles, add a line somewhere (it doesn't really matter where) which reads `; {material_bed_temperature_layer_0} {material_print_temperature_layer_0}` - it's commented out (the semi-colon) so it won't do anything to the print, but it should satisfy Cura's check to see if they exist.

Should neither of these be desirable solutions, install the Printer Settings plugin and un-check the two `Wait for <X> Heatup` settings. This should stop it adding the commands whether the placeholders exist or not; though I've not personally experimented with this.

Jak7ob, 2022-12-21:
Did this change in the last several versions of Cura? As my initial bug report stated, this was not an issue in version 4.6.1 and function as I stated. I will attempt the stated fix and see if there is any functional differences in how the Gcode generation works. 
The reason I had the temperatures hard coded, this allowed any type of material or profile to be loaded and still have the printer preheat with the calibration function running. It saved approximately one minute during a cold start and allowed me to ensure both temperatures were rising correctly.

Asterchades, 2022-12-21:
> Did this change in the last several versions of Cura?

Honestly I feel like it must have. If you'd asked me about a year back I would have sworn that it was only looking for the presence of M109 and M190 to decide whether to add its own temperature commands, but I've since been corrected. I couldn't even begin to guess which version actually made the change or why, either.

It's only the final temperatures which I was wondering for as to why they're hard-coded. You can leave the calibration temperatures fixed without causing any issues. So the code I'd probably use, based on your first example (which I imagine is what you want to keep using):

> M104 S140 ; set extruder no-ooze temp
M140 S40  ; set bed PINDA warmup temp
G21 ; set units to millimeters
G90 ; use absolute positioning
M83 ; use relative distances for extrusion
M300 S40 P10 ; chirp
G28 W ; home all without mesh bed level
G80 ; mesh bed leveling
M104 S{material_print_temperature_layer_0} ; set extruder temp
M140 S{material_bed_temperature_layer_0} ; set bed temp
M190 S{material_bed_temperature_layer_0} ; wait for bed temp
M109 S{material_print_temperature_layer_0} ; wait for extruder temp
M300 S25 P10 ; chirp
G1 Z0.250 F7200.000 ; go outside print area
G1 X50.0 E80.0  F1000.0 ; intro line
G1 X160.0 E20.0  F1000.0 ; intro line
G1 Z0.200 F7200.000 ; intro line
G1 X220.0 E13 F1000.0 ; intro line
G1 X240.0 E0 F1000.0 ; intro line
G92 E0.0 ; reset extruder distance position
M82 ;absolute extrusion mode

This will cause it to pre-heat to 140/40 for calibration, then heat to the temperatures as specified by the material properties or print profile (the latter takes priority), rather than having it try to heat _every_ material to the same 212/60 without manually editing the code. _Note that I feel there's at least one superfluous command and one superfluous parameter in there, but I've left them in tact as I'm not confident enough in the differences between RRF and Marlin or Klipper to know whether they're needed or not._