Ultimaker/Cura #14084: Model Slicing Failure

Xa-Z-aX, 2022-12-19:
### Application Version

5.2.1

### Platform

Windows 11

### Printer

Custom FFF

### Reproduction steps

1. Open STL file.
2. Scale size to 50000% (imports in weird scale from Cinema 4D)
3. Use a 0.15mm with custom temp and infill density (204° and 15% respectively)
4. Slice. Slice then fails
[CFFFP_laso master skull with base.zip](https://github.com/Ultimaker/Cura/files/10255394/CFFFP_laso.master.skull.with.base.zip)
[cura.log](https://github.com/Ultimaker/Cura/files/10255466/cura.log)

### Actual results

Just get an error to report a bug

### Expected results

i get a fully sliced object for print.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

N/A

MariMakes, 2022-12-19:
Hey @Xa-Z-aX,

Welcome to the Ultimaker Cura Github 🚀 
Thanks for your report 👍

[image]

This mesh seems faulty
Have you tried doing a quick mesh fix with the Meshtools Plugin? 
You can get it here: https://marketplace.ultimaker.com/app/cura/plugins/fieldofview/MeshTools
If that doesn't fix it you can try a more elaborate mesh fixer like https://myminifactory.github.io/Fast-Quadric-Mesh-Simplification/.

To me, it doesn't seem like a bug in Ultimaker Cura, since the mesh is faulty and Cura is not built to alter meshes in this way, that's what CAD software is for. 
Would you still like to keep this issue open or do you agree that we can close it?

Xa-Z-aX, 2022-12-20:
Thanks for the help, yeah i figured there would be some meshing issues but not enough to cause it to fail.
We can close.

closed 2022-12-20 by MariMakes