Ultimaker/Cura #14085: Slicer does not identify internal bridges

joegatling, 2022-12-19:
### Application Version

5.2.1

### Platform

MacOS Ventura 13.1

### Printer

Anycubic Kossel

### Reproduction steps

1. Load the attached STL file
2. Slice using the attached settings.
3. Enter preview and observe the orientation of the bridges on the features on the underside

### Actual results

[image]

The L-shaped bridge the touches the edge of the model has bridges in a logical orientation. However the internal bridges (that don't reach the edge) are not oriented to produces the shorter possible bridge.

### Expected results

All bridges should be oriented to span the shortest possible distance.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

The original 3D model can be found here: https://cad.onshape.com/documents/eb3e7724a7f379cb4e99416d/w/e6a478e4d0c2b9919ff2cac4/e/89be896c0013e8e5cdf3c520?renderMode=0&uiState=639fd845279463362d15f56d

MariMakes, 2022-12-22:
Hey @joegatling,

Welcome to the Ultimaker Cura Github 🚀 

I have a feeling that this might have to do with the size of of [Bridge Skin Support Threshold](https://github.com/Ghostkeeper/SettingsGuide/blob/master/resources/articles/experimental/bridge_skin_support_threshold.md) 

I tried to download your model from Onshape but I couldn't find the download button 😢 
Do you have a project file for us? It contains the printer and settings we need for troubleshooting. 
To save a project file go to File -> Save project. Please make sure to .zip your project file. For big files, you may need to use WeTransfer or similar file-sharing sites.

You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser.

joegatling, 2022-12-23:
Hi @MariMakes, I have updated the original post with the project file and also updated the sharing setting for the OnShape project to allow exporting.  I'll also have a play with the skin support threshold and see what I can see.

Thanks for helping out!

joegatling, 2022-12-23:
Attached are the results of changing that value to various extremes. I can't seem to make it to recognize the internal bridges, although I can break the detection of the one good bridge. In each screenshot the value for **Bridge Skin Support Threshold** is visible.

[image]
[image]
[image]

MariMakes, 2022-12-23:
Hey @joegatling,

I took another look, now with your project file.

You are correct **Bridge Skin Support Threshold** is not the setting that needs adjusting here. 
Instead, you can alter the top/bottom line directions. You'll get a result like this:
[image]
[Mari_AKP_BridgeTest.zip](https://github.com/Ultimaker/Cura/files/10293965/Mari_AKP_BridgeTest.zip)

To me, it doesn't seem like a bug in Ultimaker Cura, since we have a solution in place. 
Would you still like to keep this issue open or do you agree that we can close it?

joegatling, 2022-12-23:
Hi @MariMakes. Thanks for the tip on the **Top/Bottom Line Direction** feature, I was not aware of this. 

However in playing around with this feature I have found two additional issues. If you would like me to close this particular issue and open new ones, since these are not technically the same issue, please let me know.

Firstly, the direction setting seems to be getting ignored by that L-shaped feature. Notice that it initially adheres to the specified direction of 0º, but then on the next layer reverts to printing 45º lines. 
[image]
[image]

Secondly, when entering a list of directions to produce alternating lines (eg`[0,45,135]`), the surface will start at an arbitrary element in the list. (I suspect it is not actually arbitrary, instead it is using `layer index % length of array` to determine which direction to use). This would make it very hard to use this feature to reliably have the shortest possible bridge be selected in all cases.

MariMakes, 2022-12-29:
Hey @joegatling,

Thanks for the update.

With regards to the angles, that is where the originally mentioned [Bridge Skin Support Threshold](https://github.com/Ghostkeeper/SettingsGuide/blob/master/resources/articles/experimental/bridge_skin_support_threshold.md) can start playing a role. 
[image]

I'll have to double-check your remark regarding the arbitrary element. Since this setting is not specifically for bridges but for all top and bottom layers. I suspect as well that it will start at layer 1 with the suggested direction and cycle through the array.