Ultimaker/Cura #14088: failure to slice

critchy1, 2022-12-19:
### Application Version

5.2.1

### Platform

Windows 10

### Printer

Ultimaker S5

### Reproduction steps

1.sliced a model 
2. mirrored it 
3. imported another model (1)
4. imported another model (2)
5. mirrored model import (2)
6. rearranged them on build plate
7. sliced in 0.3 fast draft mode with 5mm brim 
8. error as pictured appears
9. changed brim to 7mm
10. sliced as normal
11. no error occurs

### Actual results

[image]

### Expected results

a printable sliced model should be available

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/10260067/cura.log)
[cylinder valve hrt.zip](https://github.com/Ultimaker/Cura/files/10260075/cylinder.valve.hrt.zip)

GregValiant, 2022-12-20:
Thanks for the report and the files
This is your model in the Cura preview screen.   All of those red areas are errors in the models.
[image]

The Mesh Tools Plugin is available in the Cura market place.  Load it, select all models, use the Mesh Tools / Fix Model Normals command.  Then your models will look like this and should slice.

[image]

[image]

I'll leave this open for now but it looks like a model error rather than a Cura bug.  Let us know if repairing the models allows your slice to complete.

critchy1, 2022-12-20:
Hi Greg, I'm still fairly new to slicing and printing, this is something I had no idea about ! I did as you suggested and it slices just fine 😃Thanks for the lesson, I will be sure to use this again in future . Many thanks, Paul.

critchy1, 2022-12-20:
That's great advice and very much appreciated 😁