Ultimaker/Cura #14092: Cura Slicing Error

cowdemon, 2022-12-19:
### Application Version

5.21

### Platform

Windows 10

### Printer

Creality Cr-10s Pro

### Reproduction steps

1. Attempted to slice a file using standard quality default settings.
2. Progress bar for slicing when to about ~50% before failing.
3. Greeted with error message "Slicing failed with unexpexted error. Please consider reporting a bug on our issue tracker."
4. Attempted to realign print, change preset quality/size and mode alignment.
[image]
5. No other information was given to try to resolve issue.

### Actual results

Continued error message.

### Expected results

File slicing.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

NA

Hibbary, 2022-12-20:
Same bug, except my printer is Creality Ender 3 v2.

Sonicboom247, 2022-12-22:
same bug, windows 10 Artillery Sidewinder X1 if it matters

MariMakes, 2022-12-22:
Hey @cowdemon,

Welcome to the Ultimaker Cura Github 🚀 
Thanks for your report 👍

Do you think it could be related to https://github.com/Ultimaker/Cura/issues/13643
Where the model does slice if you rotate it and move it away slightly from the center of the build plate? 

Do you have a project file for us? It contains the printer and settings we need for troubleshooting. 
To save a project file go to File -> Save project. Please make sure to .zip your project file. For big files, you may need to use WeTransfer or similar file-sharing sites.

You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser.

closed 2023-01-06 by github-actions[bot]