Ultimaker/Cura #14095: Cura slicer printing purge tower layers in air for 2nd colour

Sednav, 2022-12-20:
### Application Version

5.2.1

### Platform

Windows 11

### Printer

Creality CR-X

### Reproduction steps

1. Created 2 test models with 2 PLA colours; each with white and clear (I had already printed these models with rough settings using the default Creality slicer; see attached image)
2. combined the 2 models to print together on a raft base.
3. After slicing the combined model and inspecting the layers I saw an overhang step inside the tower where the 2nd colour started, from layer 103 to 104
4. Couldn't find any suitable support settings so began to print this model on a raft base, including purge tower
5. I had to attend to Christmas shopping and let it print continue while away.
[image]

### Actual results

6. When I returned, the model was a mess of loose, tangled strands everywhere; a virtual bird's nest.
7. Apparently, when the 2nd colour started, the strand purged into new circles inside the tower, like the preview animation showed.
8. The first circle started printing 3 layers inside the tower inner wall, but just strung down onto the raft, quickly creating a birds nest.
9. Subsequent layers then caught the tangled birds nest strands and pulled them outside everywhere
10. Model was ruined and I stopped it.

[image]

### Expected results

I expected the clear and white colours to print side-by-side, according to the model designations. The purge tower was meant to keep each colour clean without streaking and colour bleeds. It didn't get that far because of the unsupported layers and subsequent purge tower fail.

I had printed part of this 2-colour model reasonably successfully using the Creality slicer (apparently an early Cura edition), in rough, fast preliminary mode, and wanted a more accurate print with Cura.

[CCRX_04_08_BCHBB_05.zip](https://github.com/Ultimaker/Cura/files/10266444/CCRX_04_08_BCHBB_05.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

I did a Google search for this problem and found the following links, but without an actual solution:

https://github.com/Ultimaker/Cura/issues/6909

https://www.google.com/search?q=cura+prime+tower+colors+stacked&tbm=isch&ved=2ahUKEwiKwoCMuYf8AhXjhNgFHRlgChcQ2-cCegQIABAA&oq=cura+prime+tower+colors+stacked&gs_lcp=CgNpbWcQA1AAWMsmYOYqaAFwAHgAgAGEAogB5BOSAQYwLjE0LjGYAQCgAQGqAQtnd3Mtd2l6LWltZ8ABAQ&sclient=img&ei=rkWhY4qzF-OJ4t4PmcCpuAE&bih=760&biw=1536#imgrc=VcPiyWv8HDHBxM

Please help me get this model printed.
Merry Christmas

Cheers, Sednav

Soprano80, 2022-12-20:
[image]

here the error in the Cura view, you can see the 2nd color green is in the air

Sednav, 2022-12-21:
After some thought, I could just construct a bespoke purge tower easily enough as part of the model, but without painstakingly hacking the G-code, I can't see an efficient way to actually start and finish a new colour ON the prime tower - which is its purpose. If you could just point me in the right direction with that, the problem could easily be solved. (I am printing PLA in both colours, so there is no compatibility issue)

Sednav, 2022-12-22:
I found that if the purge tower is increased in diameter to 35mm, and wipe toggled on, the dropped strands inside it have more space to settle and don't seem to get dragged out. I had one glob forming inside the tower, which was eventually dragged out, but I was watching at the time, and could pause the print, snip the glob off, and remove it, before resuming the print. Finished print quality was much worse than with using the Creality slicer in rough mode, so I still have to fiddle around with settings for a while.

Sednav, 2022-12-22:
Here is the current result of the test print, mainly to test tower settings, and using a brim instead of a raft. The model is not very high.

MariMakes, 2022-12-22:
Hey @Sednav,

This looks like such an ambitious project 😮 

My first instict would be to turn around your hull so it prints the support internally, I expect that you will be looking at the outside of the boat anyways so you want that to look the best! 

You can also place a small mesh next to the prime tower if you think tha tyour printjob fail might be related to purging. Something like this:
[image]

I think your fail might be related to the fact that your mesh is faulty. 
Have you tried doing a quick mesh fix with the Meshtools Plugin? 
You can get it here: https://marketplace.ultimaker.com/app/cura/plugins/fieldofview/MeshTools

Because the model is faulty your support is being printed inside of the model causing a build up of material that is probably causing the fail. 
[image]

Do you own the CAD model?
Would it be possible to alter it to make it solid or at least the walls thicker?
Could you check the export settings of your CAD software to make sure that the walls are watertight? 
If you print with these thinner walls I really suggest to look into the specific Arachne Engine settings that we introduced in Cura 5.0. You can read more about them here: https://support.ultimaker.com/hc/en-us/articles/4792077687068

If you want try it with tree support, you can try to use the newly released Alpa with a new version of tree support. 
you can download it here. https://github.com/Ultimaker/Cura/releases/tag/5.3.0-alpha%2Bxmas

To me, it doesn't seem like a bug in Ultimaker Cura, since the mesh is faulty and Cura is not built to alter meshes in this way, that's what CAD software is for. 
Would you still like to keep this bug open or do you agree that we can close it?

Sednav, 2022-12-23:
Hi MariMakes,
Thank you for your comprehensive response. It is very much appreciated. I will respond to your questions and recommendations in sequence:
1. Ambitious for sure. Turning the model upside-down is certainly possible, but my main objective is to actually depict the interior layouts of each deck so a client can imagine themselves moving around in it. Upside down would create massive amounts of overhang supports inside, which would leave unsightly connection marks on the finished floors. In this model decks can be lifted off individually to view the interior layout. I have already worked out clip and connector details to stack the decks and keep them securely aligned. Here are some images of the lower 3 decks, with 4 upper decks and mast with sails still to be modelled, as well as a cradle with CFD-generated wave train at 20 knots (I have already modelled part of the wave train, which works quite well).
[image]
[image]
[image]

[image]

2. Your extra mesh tower looks like a great idea. I have meanwhile doubled the prime tower volume to differentiate the clear material but only the outer layers (extruder 1) is increased but the clear is still far too contaminated with white to be useful so I will double the volume again and again, until reasonably clear. Your extra tower idea should purge more as well, as long as that is the next item printed after the nozzle change and purge tower layer.
3. Yes there are flaws in that particular model, and I have meanwhile remodelled and extended it to be a perfectly closed manifold mesh, and printed it again. I have attached the corrected and extended model.

[image]
[image]
4. Yes, I created this comprehensive model in Rhino3D and own it. I am a naval architect (now retired), with extensive 3D modelling and CNC experience since the later 70s, and have designed this ship outright. In the 70s my first computers came with an operating system (firstly CPM and later DOS), and a programming language (Various generations of interpreted, and then compiled BASIC), so I wrote all my early naval architecture programs myself :-). Subsequently professional IT developed and I gradually switched to their far better products. You can read more about this on my website if you want at http://www.sednav.com/spectrum-98/
5. I have already made the walls thicker than their scaled dimensions, and endeavour to keep it all as close to reality as possible. My thinnest interior walls are 0.8mm (2 time 0.4..), roughly equivalent to built walls at 1:200 scale, and outside shell is thicker to be roughly equivalent to the metal hull shell framework plus lining. Glass is of course massively exaggerated at 0.8mm, but that's just what it has to be at this scale.
6. The attached new model file (zip) was created in Rhino3D and tested for integrity.
7. I will check out the Arachne Engine settings as you recommended. I'm always eager to learn and understand more..
8. Your new tree support looks great and I will spend some time with it.
9. If the purge tower was designed for the 2nd nozzle to just purge unsupported into the inside space, then if that works ok, it is not a bug. It might be worth checking if increasing the purge volume only affects the 1st nozzle or not. I can't really tell. But for my application, with laterally alternating 1st and 2nd colour strips  I am pretty certain I need far more purge volume to differentiate the colours sharply.

Thank you again MariMakes, for your comprehensive help. This has set me on the right path for this year-long, challenging project.
Best Regards, Merry Christmas and Happy New Year, and may all your hopes and dreams come true.

Sednav, 2022-12-23:
HI MariMakes,

Latest update:

1. I've downloaded UltiMaker Cura 5.3.0-alpha+cura_9879_19 for the extra tree support and it works.
2. I have attempted to implement your suggestion of "...a small mesh next to the prime tower..." as you suggested, but I can't get this to print FIRST, immediately after the purge tower. It persistently prints AFTER printing the model parts, which is pointless. I have searched documentation and found nothing to help. Please show me how to set this print sequence correctly.
3. Here is the new file with this extra tower, which needs to print FIRST for the 2nd extruder purge to work.

Looking forward to your solution.
Cheers, Sednav.

[CCRX_08_BCHF_08.zip](https://github.com/Ultimaker/Cura/files/10292554/CCRX_08_BCHF_08.zip)

MariMakes, 2022-12-23:
Hey @Sednav,

Thanks for the updates. 
I'm happy to see you are making so much progress.

You can alter the layer start settings to change the order in which the models are printed so you can do the Purge tower first. 
[image]

I changed the status of the ticket from a bug to a discussion so other people can help you in your journey. Since this is a place where we track bugs and feature requests. 

You can also consider starting to post your findings on https://community.ultimaker.com/ where even more people can help you on your 3D printing journey.

Sednav, 2022-12-23:
Thank you MariMakes

That works in the animation, so I'll try it in the modal as well now.
I will start posting in the Community next time.

Merry Christmas.