Ultimaker/Cura #14096: Slicing part cone

Sanje67, 2022-12-20:
### Application Version

5.1.0

### Platform

Windows 10

### Printer

Ultimaker S5

### Reproduction steps

trying to get a stl part ready for printing. Just pushing the button slice. At the end of the proces a error occured and send me to Github

### Actual results

process stops

### Expected results

Cura should calculate the part forthe necessary layers and supports

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[UMS5_cone half v2.zip](https://github.com/Ultimaker/Cura/files/10267191/UMS5_cone.half.v2.zip)

GregValiant, 2022-12-20:
Thanks for the report.
Moving and rotating the model didn't get it to slice, but changing the "Initial Layer Height" and the "Layer Height" both to 0.28 did allow it to slice.

Someone from the Cura team will take a look.
[image]

jellespijker, 2024-01-14:
Thank you for your report. It seems that your uploaded project file slices correct on our latest unreleased 5.7.0-alpha.0+3abb5 which indicates that the issue you're experiencing seems to be fixed for the upcoming 5.7.0 release.
I'm closing this issue as fixed for the 5.7.0 release

[image]