Ultimaker/Cura #14097: Camera view in UI is locked. Can´t zoom, rotate, pan nothing.

tascheri, 2022-12-20:
### Application Version

5.2.1

### Platform

windows 10

### Printer

custom printer

### Reproduction steps

Open the program
View is locked. not even the buttons on the lower left work.

### Actual results

the view is always the same 
[image]

### Expected results

able to change view to the object

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

none

fieldOfView, 2022-12-21:
Please post a link yo your Cura.log log file. We can not help you without it.

If the Cura user interface still starts, you can also reach these directories from the application menu:

Help    »    Show settings folder  
`%APPDATA%\cura\<Ｃｕｒａ Ｖｅｒｓｉｏｎ>\cura.log`
or
`C:\Users\<your username>\AppData\Roaming\cura\<Ｃｕｒａ Ｖｅｒｓｉｏｎ>\cura.log`

MariMakes, 2022-12-23:
Hey @tascheri,

Welcome to the Ultimaker Cura Github 🚀 
Thanks for your report 👍

I recognize your issue. If you are stuck like this, sometimes you can escape by pressing Esc or clicking on the different views in the bottom left corner of your screen. 
[image]

This seems to be a duplicate of https://github.com/Ultimaker/Cura/issues/13514
We are still collecting steps to reproduce and logs so we can fix the issue.
I'll be closing this issue as a duplicate but you can follow the progress here: https://github.com/Ultimaker/Cura/issues/13514

closed 2022-12-23 by MariMakes

tascheri, 2023-03-04:
> Hey @tascheri,
>
> Welcome to the Ultimaker Cura Github 🚀 Thanks for your report 👍
>
> I recognize your issue. If you are stuck like this, sometimes you can escape by pressing Esc or clicking on the [...]

clicking on the buttones doesnt help.

tascheri, 2023-03-04:
> Please post a link yo your Cura.log log file. We can not help you without it.
>
> If the Cura user interface still starts, you can also reach these directories from the application menu:
>
> Help    » [...]

[cura.log](https://github.com/Ultimaker/Cura/files/10889619/cura.log)

tascheri, 2023-03-04:
Running 5.3.0 now, the issue persists...

malcolm-whitlock, 2023-03-30:
I've had the same problem since version 5.0.0 and none of the updates have worked however I have just found a solution on another post that worked for me on version 5.3.0. To solve this go to the 'marketplace' (in the top right of the menu bar), click on the 'gear wheel' and enable 'camera tool'.

diabeetus01, 2023-11-05:
This has now started happening for me on Windows in Cura 5.5.0. There is no longer a "Camera Tool" plugin showing up under the Settings cog as described above, though I can find said tool in the install files.

Reverting back to 5.4.0 the Camera Tool shows up and is enabled, however clicking to drag etc does nothing.
I just went from Linux to Windows on my laptop to see how things differed, but didn't have such issues on the Linux version. Strange behavior, will probably switch back and see if it's working on the Linux version soon, and put up with the slow scrolling speed there for settings.