Ultimaker/Cura #14099: Reseting Z-Axis

icegeiser, 2022-12-20:
### Application Version

5.2.1

### Platform

Windows 10

### Printer

Ultimaker 

### Reproduction steps

When i do a slicer in some 3D models i see that the Z-axis goes to home position, on the G-code.

### Actual results

Results:
"
G1 F900 X85.33 Y80.415 E4156.79367
G0 F7200 X84.761 Y80.418
G1 F900 X93.389 Y71.79 E4157.09805
G0 F7200 X93.318 Y71.296
G1 F900 X84.182 Y80.431 E4157.42033
G0 F7200 X83.619 Y80.429
G1 F900 X93.241 Y70.806 E4157.75979
G0 F7200 X92.972 Y70.51
G92 E0
G1 F900 X83.077 Y80.405 E0.34907
G0 F7200 X82.536 Y80.381
G1 F900 X92.703 Y70.214 E0.70774
G0 F7200 X92.468 Y69.883
G1 F900 X81.994 Y80.357 E1.07724
G0 F7200 X81.578 Y80.207
G1 F900 X92.284 Y69.501 E1.45492
G0 F7200 X92.1 Y69.119
"

### Expected results

Expected:
"
G1 F900 X85.33 Y80.415 E4156.79367
G0 F7200 X84.761 Y80.418
G1 F900 X93.389 Y71.79 E4157.09805
G0 F7200 X93.318 Y71.296
G1 F900 X84.182 Y80.431 E4157.42033
G0 F7200 X83.619 Y80.429
G1 F900 X93.241 Y70.806 E4157.75979
G0 F7200 X92.972 Y70.51
G92 E0  => NOT WITH THIS LINE.
G1 F900 X83.077 Y80.405 E0.34907  => This extruder must continue the before value + increment.
G0 F7200 X82.536 Y80.381
G1 F900 X92.703 Y70.214 E0.70774
G0 F7200 X92.468 Y69.883
G1 F900 X81.994 Y80.357 E1.07724
G0 F7200 X81.578 Y80.207
G1 F900 X92.284 Y69.501 E1.45492
G0 F7200 X92.1 Y69.119
"

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[CFFFP_Larynx_Model_BodyParts_3D_and_University_of_Dundee_School_of_Medicine_Share_Alike_License.zip](https://github.com/Ultimaker/Cura/files/10270647/CFFFP_Larynx_Model_BodyParts_3D_and_University_of_Dundee_School_of_Medicine_Share_Alike_License.zip)

GregValiant, 2022-12-20:
Thanks for the report.
I assume you mean the "E" value.  If the "Z" went home the print would get smashed.

Constant calculation of the E value results in cumulative errors due to rounding.  Every so often Cura will reset the E to 0 to cancel the errors that have crept in.  With 1.75 filament that happens as the E approaches 4158mm (10,000mm³).  The same number may be used for 2.85 filament.
There is no need for a reset when using "relative extrusion" mode as the E value is reset on every line.

I'm going to remove the bug label as this is expected behavior.  I will leave this open in case a member of the Cura team would like to comment.  Let us know if you feel it should remain a bug report.

icegeiser, 2022-12-20:
I made with "relative extrusion" and the problem was not see anymore.
Thanks a lot.

closed 2022-12-20 by icegeiser

GregValiant, 2022-12-22:
My experience is that whether you use Absolute or Relative the prints are the same.  Using Relative Extrusion does make the gcode easier to read.
If a printer has a default of Relative Extrusion and doesn't understand M82 then there would be problems.
The opposite is also true.  If the firmware couples M82 and M83 with the G90/G91 Relative/Absolute movement commands then Relative Extrusion would not work as the only way to set it would be to have the entire gcode file in "Relative Movement".