Ultimaker/Cura #14105: (5.2.1) "multiply selected" won't multiply more than 99x

ungutknut, 2022-12-21:
### Application Version

5.2.1

### Platform

Windows 10

### Printer

any

### Reproduction steps

1) Add some object that would fit more than 100 times on your buildplate.
2) Hit ctrl+m
3) Enter any number >99 and click ok

### Actual results

Cura only multiplies the object 99 times max.

### Expected results

Cura should obey to the given number. 

Or at least give me a warning or prevent me from entering an invalid number. Previous versions (e.g. arachne alpha 2) only accepted 2-digit numbers which was not good but better since it didn't leave me with unexpected surprises.

Desired behaviour would be to just let me decide how many objects I want to have on the buildplate. In this case I'd like to have 200. So I have to multiply 99x, then again 99x and then again 1x which is not very intuitive.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]

MariMakes, 2022-12-23:
Hey @ungutknut,

Thanks for your report 👍
Good find! 🔎 

I can reproduce it to I agree that it should block or warn you if you get limited amount of copies. 
[image]

That bug probably sneaked in with upgrading QT in Cura 5.0.
I'll bring it up to the team to see what they can do to improve it. 
But I expect that if it's not a quick fix that it will not get a high priority to implement. 
Fingers crossed 🤞 

On another note. Why do you need 200 tiny boxes? What project are you working on? 🤩

ungutknut, 2022-12-23:
> On another note. Why do you need 200 tiny boxes? What project are you working on? star_struck

Haha... I just replaced the original geometry with blocks for the bug report since the original parts are under disclosure. 200 tiny blocks indeed would qualify for a peculiar project ^^

Thanks for investigating.

GregValiant, 2022-12-23:
Going back a few versions it appears that the limit has always been 99.  In the rework for QT6 the text box code is missing the maximum value of 99 which is being enforced elsewhere in the code.

Joeydelarago, 2022-12-30:
[CURA-10096](https://ultimaker.atlassian.net/browse/CURA-10096) ticket reference for devs.

MariMakes, 2023-10-11:
👋 Quick update on our side. 

This is resolved in the 5.5 Beta release 🎉 , you can download the Cura version with the fix here: [https://github.com/Ultimaker/Cura/releases/tag/5.5.0-beta.1](https://github.com/Ultimaker/Cura/releases/tag/5.5.0-beta.1)
I'll close this issue since it's resolved. 

Thanks again, and please let us know if you run into any other issues 💪

closed 2023-10-11 by MariMakes