Ultimaker/Cura #14106: Cura freezes in monitoring screen on long print

marcelrv, 2022-12-21:
### Application Version

5.2.1

### Platform

Windows 11

### Printer

Ender3

### Reproduction steps

Cura is freezing midway of the print.
This is while I'm in the monitor screen.
Note using the octoprint plug-in for sending the file to octoprint. The print itself is not harmed by the freezing of Cura (it is just annoying as it needs to be forced closed etc)

It happens also when the print is already running (meaning it is not dependent on the  stl that is sliced or printed)

See here a log of last 10min  until the point of freezing. It is pretty consistent happening for longer prints. (e.g. in example I believe it happened after approx 2h

```

2022-12-21 15:28:56,688 - DEBUG - [MainThread] UM3NetworkPrinting.src.Cloud.CloudOutputDeviceManager._onGetRemoteClustersFinished [169]: Synced cloud printers with account.
2022-12-21 15:29:58,178 - DEBUG - [MainThread] cura.OAuth2.AuthorizationHelpers.getAccessTokenUsingRefreshToken [64]: Refreshing the access token for [https://account.ultimaker.com]
2022-12-21 15:29:58,178 - DEBUG - [MainThread] cura.OAuth2.AuthorizationService.refreshAccessToken [174]: Was already busy refreshing token. Do not start a new request.
2022-12-21 15:29:58,178 - DEBUG - [MainThread] cura.OAuth2.AuthorizationService.refreshAccessToken [174]: Was already busy refreshing token. Do not start a new request.
2022-12-21 15:29:58,288 - DEBUG - [MainThread] UM3NetworkPrinting.src.Cloud.CloudOutputDeviceManager._onGetRemoteClustersFinished [169]: Synced cloud printers with account.
2022-12-21 15:29:58,335 - DEBUG - [MainThread] cura.OAuth2.AuthorizationService._storeAuthData [289]: Attempting to store the auth data for [https://account.ultimaker.com]
2022-12-21 15:30:08,366 - DEBUG - [MainThread] cura.AutoSave._onTimeout [64]: Autosaving preferences, instances and profiles took 0.015242815017700195 seconds
2022-12-21 15:31:00,909 - DEBUG - [MainThread] UM3NetworkPrinting.src.Cloud.CloudOutputDeviceManager._onGetRemoteClustersFinished [169]: Synced cloud printers with account.
2022-12-21 15:32:01,665 - DEBUG - [MainThread] UM3NetworkPrinting.src.Cloud.CloudOutputDeviceManager._onGetRemoteClustersFinished [169]: Synced cloud printers with account.
2022-12-21 15:33:01,792 - DEBUG - [MainThread] UM3NetworkPrinting.src.Cloud.CloudOutputDeviceManager._onGetRemoteClustersFinished [169]: Synced cloud printers with account.
2022-12-21 15:34:02,639 - DEBUG - [MainThread] UM3NetworkPrinting.src.Cloud.CloudOutputDeviceManager._onGetRemoteClustersFinished [169]: Synced cloud printers with account.
Windows fatal exception: access violation

Thread 0x0000452c (most recent call first):
  File "threading.py", line 324 in wait
  File "zeroconf\__init__.py", line 2534 in wait
  File "zeroconf\__init__.py", line 1715 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x0000490c (most recent call first):
  File "zeroconf\__init__.py", line 1369 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x00003ff4 (most recent call first):
  File "C:\Program Files\Ultimaker Cura 5.2.1\share\cura\plugins\USBPrinting\USBPrinterOutputDeviceManager.py", line 88 in _updateThread
  File "threading.py", line 946 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x00004310 (most recent call first):
  File "threading.py", line 324 in wait
  File "zeroconf\__init__.py", line 2534 in wait
  File "zeroconf\__init__.py", line 1715 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x00005148 (most recent call first):
  File "threading.py", line 324 in wait
  File "threading.py", line 600 in wait
  File "C:\Program Files\Ultimaker Cura 5.2.1\share\cura\plugins\UM3NetworkPrinting\src\Network\ZeroConfClient.py", line 81 in _handleOnServiceChangedRequests
  File "threading.py", line 946 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x00000ac8 (most recent call first):
  File "zeroconf\__init__.py", line 1369 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x00003fd4 (most recent call first):
  File "C:\Program Files\Ultimaker Cura 5.2.1\share\cura\plugins\RemovableDriveOutputDevice\RemovableDrivePlugin.py", line 61 in _updateThread
  File "threading.py", line 946 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x000018b4 (most recent call first):
  File "UM\Backend\Backend.py", line 169 in _storeStderrToLogThread
  File "threading.py", line 946 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x00003a5c (most recent call first):
  File "UM\Backend\Backend.py", line 157 in _storeOutputToLogThread
  File "threading.py", line 946 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x00002118 (most recent call first):
  File "threading.py", line 320 in wait
  File "threading.py", line 460 in acquire
  File "UM\JobQueue.py", line 98 in _nextJob
  File "UM\JobQueue.py", line 124 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x00005494 (most recent call first):
  File "threading.py", line 320 in wait
  File "threading.py", line 460 in acquire
  File "UM\JobQueue.py", line 98 in _nextJob
  File "UM\JobQueue.py", line 124 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x0000453c (most recent call first):
  File "threading.py", line 320 in wait
  File "threading.py", line 460 in acquire
  File "UM\JobQueue.py", line 98 in _nextJob
  File "UM\JobQueue.py", line 124 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x00005bd0 (most recent call first):
  File "threading.py", line 320 in wait
  File "threading.py", line 460 in acquire
  File "UM\JobQueue.py", line 98 in _nextJob
  File "UM\JobQueue.py", line 124 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x00002f58 (most recent call first):
  File "threading.py", line 320 in wait
  File "threading.py", line 460 in acquire
  File "UM\JobQueue.py", line 98 in _nextJob
  File "UM\JobQueue.py", line 124 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x00005840 (most recent call first):
  File "threading.py", line 320 in wait
  File "threading.py", line 460 in acquire
  File "UM\JobQueue.py", line 98 in _nextJob
  File "UM\JobQueue.py", line 124 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x000008d0 (most recent call first):
  File "threading.py", line 320 in wait
  File "threading.py", line 460 in acquire
  File "UM\JobQueue.py", line 98 in _nextJob
  File "UM\JobQueue.py", line 124 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x00005b20 (most recent call first):
  File "threading.py", line 320 in wait
  File "threading.py", line 460 in acquire
  File "UM\JobQueue.py", line 98 in _nextJob
  File "UM\JobQueue.py", line 124 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Current thread 0x00004390 (most recent call first):
  File "UM\Qt\Bindings\MainWindow.py", line 284 in _render
  File "UM\Qt\QtApplication.py", line 416 in exec
  File "cura\CuraApplication.py", line 897 in run
  File "cura_app.py", line 239 in <module>

mv@MIBOOK MINGW64 /c/Program Files/Ultimaker Cura 5.2.1
$
```

### Actual results

Cura freezing after approx 2h in the monitoring screen.
I'm not sure that the time is consistent, but it happens not occasionally but frequently

### Expected results

Cura remains  running / responsive until closed

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura52.log](https://github.com/Ultimaker/Cura/files/10278768/cura52.log)
See here a log of the starting until the point of freezing (partly pasted above)
[debugtext.txt](https://github.com/Ultimaker/Cura/files/10278770/debugtext.txt)

MariMakes, 2022-12-23:
Hey @marcelrv,

Welcome to the Ultimaker Cura Github 🚀 
I'm sorry to hear your Cura is freezing but I'm glad that your printjob survived the ordeal.

We've had reports of people experiencing slower and freezing software after upgrading to Windows 11.
But we are having trouble with reproducing it in house, this seems like something we haven't tried yet. Thank you!
We've been collecting them here https://github.com/Ultimaker/Cura/issues/13820

I'll bring it up to the team to see what they think. 
Fingers crossed 🤞

MariMakes, 2023-01-10:
Hey @marcelrv,

I brought your issue up to the team. They are really curious about your case. 
To learn better about what's slowing your software down, could you install a plug-in that performs more extensive logging? 

You can download it here: https://marketplace.ultimaker.com/app/cura/plugins/UltimakerPackages/ExtensiveSupportLogging
After it's installed can you... 
	1. Select Extensions
	2. Select Remote Support
	3. Start CPU profiler
	4. Start a short printjob
	5. Stop CPU profiler
	6. Select Save Remote Support Archive
	7. Go to the Marketplace
	8. Select the gear icon 
	9. Uninstall Startup Optimiser
	10. Share the Remote Support Archive Zip file here on Github

You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser. 

That would really help us out in troubleshooting what you, and others, are reporting.

closed 2023-01-25 by github-actions[bot]