Ultimaker/Cura #14107: combining fails if wall thickness changes

UBaer21, 2022-12-21:
### Application Version

5.2.1

### Platform

Windows 10

### Printer

FDM

### Reproduction steps

slicing object

### Actual results

combining failed i certain areas
line set to 0.21
random seam:
[image]
shortest seam
[image]

line set to 0.2 (random seam)
[image]

### Expected results

nozzle is kept inside print and not traveling (and retracting). 
Setting a longer Max Comb Distance With No Retract causes travels through the center without retractions.

in other layers this is working 
[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

seems slight changes in the object wall thickness causing these travels

MariMakes, 2022-12-23:
Hey @UBaer21,

Welcome to the Ultimaker Cura Github 🚀 
Thanks for your report 👍

At first glance your analysis seems correct, if the walls are too thin there is nowhere to comb towards.
Did you consider decreasing the minimum wall line width?
Also, I'm super curious, why do you prefer the random seam for your print job? 

Do you have a project file for us? It contains the printer and settings we need for troubleshooting. 
To save a project file go to File -> Save project. Please make sure to .zip your project file. For big files, you may need to use WeTransfer or similar file-sharing sites.

You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser.

UBaer21, 2022-12-23:
[CuraWalltest.3mf.zip](https://github.com/Ultimaker/Cura/files/10294704/CuraWalltest.3mf.zip)

Please find attached a project file to demonstrate the problem.  As 3mf is already a zip file just remove the .zip ending.
I like to point out that Cura created a geometry full of defects - [the original has proper geometry](https://www.3dcustomizer.net/paste/2016B79F). 

the minimum wall line width is already decreased to .1  changes seem to have little effect here. 
The old 4.x version didn't had this problem as long walls been over 2×line wide,  interesting is that even with a wall count of 1 (so there is enough space between) this is happening. Also if choosing "shortest seam":
[image]
also it is less prominent. 

Only option is using custom seam that eliminate the problem. 

Regarding random seam.  This is the only way to have an invisible seam as all other options align the seam which is causing minimal stringing to build up at this location which is causing small gaps to fuse (print in place)  also the print will not be symmetrically.  And if using small features defects (retraction/travel/start/end)  always align at the same position  which is causing problems to accumulate till a point that walls get holes. With random these are distributed and didn't cause that much of an issue.

MariMakes, 2022-12-23:
Hey @UBaer21,

Thanks for the project file. 
I came to the same conclusion as you that shortest is the best solution if you want the seam to be more random. 

I'll bring it up to the team to see what they can do to improve it.
But I expect that if it's not a quick fix that it will not get a high priority to implement. 
Fingers crossed 🤞