Ultimaker/Cura #14109: Tree support inner structure floating, not connected to anything

alexanderphoenix, 2022-12-22:
### Application Version

4.13.1

### Platform

Linux (Manjaro 22.0.0)

### Printer

Flsun Q5

### Reproduction steps

The inner section of tree supports float in mid air and are not attached to anything.
Example:
[image]
Example project (3MF) and GCODE files included below.

### Actual results

Parts of the inner tree structure are floating and will not print properly

### Expected results

All parts of the tree support structure are supported from the bottom

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

Example project (3MF) and GCODE files:
[FQ5_gaggia-classic-drip-tray.zip](https://github.com/Ultimaker/Cura/files/10282604/FQ5_gaggia-classic-drip-tray.zip)

alexanderphoenix, 2022-12-22:
Not realising I was running an outdated version, I just updated to 5.2.1 and the issue still persists, though it has to be said it seems like it is trying a bit harder to give that floating piece more support.
Unfortunately as shown below, it is still disconnected and therefore it is lacking the necessary support.
[image]
[FQ5_gaggia-classic-drip-tray_cura5.2.1.zip](https://github.com/Ultimaker/Cura/files/10282701/FQ5_gaggia-classic-drip-tray_cura5.2.1.zip)

GregValiant, 2022-12-22:
Thanks for the report.
Your attached file is a 3mf model file.  A Cura project file ("File | Save Project") would contain all your settings and your printer.  I only received the model so I must guess.
There are 73 or 74 settings in the Support section and they need to be set correctly to give proper support to your model.
In this case your setting "Tree Support Branch Diameter Angle" might be inappropriate.  Try it at 4° and see how the support looks.
I'll leave the bug label on this but it is likely a settings issue.  See if making some changes to the support settings improves the support structure for that model.  Let us know how you make out.

alexanderphoenix, 2022-12-22:
Hi @GregValiant,

thank you for your reply.
It seems that on Linux Cura wants to save the project as 3MF, it calls it `Cura Project 3MF file`, and there are no other options in the dropdown menu. Do you know what extension cura projects are saved as?
[image]
Also the name never populates on Linux which is annoying, not sure if maybe this is another bug?

I'll try the settings as you suggested, thank you for that!
I would have thought floating geometry should still not occur regardless of the options I choose, however.
Cura is immensely complicated in the way calculates the Gcode so I thought it may have been useful to report this behaviour

GregValiant, 2022-12-22:
I'm not familiar with the screen.
In the Windows version there is a "File" menu at the left end of the Cura menubar.  One of the choices is "Save Project" which creates a "3mf" project file including the active printer, all the Cura settings, and the model.
The other relevant choice is "Export..." which creates a "3mf" model file that contains only the model and that's what I received.  Consequently I can't tell how you have set up the slice.

When I first sliced the model there was a support piece that just appeared and was unattached to the support structure.  Changing the "Tree Support Branch Angle" from the default 5° down to 4° seemed to fix the problem.

MariMakes, 2022-12-22:
Hey @alexanderphoenix,

Welcome to the Ultimaker Cura Github 🚀 

This seems to be a duplicate of https://github.com/Ultimaker/Cura/issues/14041
We have a ticket on our backlog with the intent to improve this behavior.

Can I close this ticket as a duplicate? 
Or is the suggestion from @GregValiant what you were looking for?

I'll add your model it as a use case regardless so we can improve our tree support 😉

alexanderphoenix, 2022-12-22:
@GregValiant apologies for not giving you context, that is the "save" prompt I get in gnome (linux desktop environment) when I click on `Save Project`. It's similar to window's one, where it forces you to save with a specific extension, in this case "3MF". I din't click on export, which is why I'm baffled by the lack of options for the project.
Thank you once again for testing the option you mentioned, I'll follow your suggestion.

@MariMakes thank you for finiding  and linking the duplicate, I wasn't completely sure what terms to look for so I did a search but didn't stumble across this one.
Please feel free to close this as it's clearly a duplicate, and thank you for adding the files I attached to the other case!