Ultimaker/Cura #14112: "arrange all models" not working

TheDeejster, 2022-12-22:
### Application Version

5.2.1

### Platform

Windows 10

### Printer

Ender 3 Pro

### Reproduction steps

1. Place multiple models in the project (seems like at least 3)
2. Arrange all models

### Actual results

Usually, one or two models are properly arranged, and the rest (even though there's plenty of space) are put off to the side and grayed out.

### Expected results

All models should be properly arranged on the build plate, assuming ample space.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]
[image]

MariMakes, 2022-12-23:
Hey @TheDeejster,

Welcome to the Ultimaker Cura Github 🚀 
Thanks for your report 👍

We know that when approaching the maximum build plate footprint our current arranging algorithm doesn't always do the job correctly. This is something we accepted during implementation because it's not often that people use their full build volume and if they do there are often enough options to correct by hand.

I'll bring it up to the team to see what they can do to improve it. But I expect that if it's not a quick fix that it will not get a high priority to implement. Fingers crossed 🤞

MariMakes, 2022-12-30:
Hey @TheDeejster,

I brought your issue up to the team. They mentioned that another reason might be that the arranging algorithm doesn't make 45-degree turns, only 90 degrees because else it would get even slower than it already is. 

We believe we will not work on this anytime soon and will therefore defer this issue.
Since Cura is open source, you or anyone else is more than welcome to work on this issue and create a pull request yourself.

I hope you understand 🙏

TheDeejster, 2022-12-30:
While this isn’t the issue, as I am able to easily arrange the models on
the build plate myself without making any turns, I do understand that the
issue is not high priority. I’m just glad someone is actually reading the
reports. Thank you!

tnemrap, 2023-01-14:
I have the same problem even with 90 degrees turns and have to arrange them manually.

Knaudler, 2023-02-15:
Hello
same Problem
Using Ultimaker Printer for Business
Cura 5.2.2 - we need this function
in older Versions it works, but now I have to arrange all Models all time with hand ?
Please fix

MariMakes, 2023-02-15:
Hey @Knaudler,

I believe we haven't touched the arranging algorithm in quite a while. 
Do you have an example of "in older Versions it works?"
Do you have a file which worked before, but doesn't work anymore in 5.2.2?

Knaudler, 2023-02-15:
hello

I have to check, old cura versions deleted, today I changed to prusa
slicer...

MariMakes, 2023-03-20:
Hi, Update on our side. 

Our developers spotted an error where there is way too much emphasis on an object being in the center of the buildplate rather than making sure that all the objects fit. 

We've added a ticket to the backlog with the intent to improve this. 
For internal reference CURA-10403

MariMakes, 2023-10-24:
Quick update from our side 👋

In the 5.5 beta we introduced a grid arrange that helps here, but during the 5.5 beta, we spent some time improving the arrange for larger models and introduced an approach that we think might resolve this issue. 

You can expect a fix in the 5.5 stable soon ™️ 

Thanks for your report 👍

closed 2023-10-24 by MariMakes