Ultimaker/Cura #14113: (5.3.0) Slicing with supports failed

truegret, 2022-12-22:
### Application Version

UltiMaker-Cura-5.3.0-alpha+xmas

### Platform

Windows 11

### Printer

Creality 10S

### Reproduction steps

1. Launch Cura 5.3.0-alpha-xmas
2. Load/Import/Open existing .STL (note: slices/prints OK with Cura5.2.1, same profile)
3. Clicked Slice button (automated slicing is disabled) 

### Actual results

Roughly midway (40-50%) through the slicing process I receive a Slicing error popup message.

[image]

### Expected results

A completed slice and gcode generated.  The same profile and model successfully slice (and print) with UltiMaker-Cura-5.2.1.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

A zip file has been added ( UltiMaker-Cura-5.3.0-alpha-xmas SlicingBug.zip ).  This file includes the project file, log file, and screenshot of bug popup.

[UltiMaker-Cura-5.3.0-alpha-xmas SlicingBug.zip](https://github.com/Ultimaker/Cura/files/10289633/UltiMaker-Cura-5.3.0-alpha-xmas.SlicingBug.zip)

MariMakes, 2022-12-23:
Hey @truegret,

Thanks for being the first report from our Christmas Alpha! 🎄 🎉 

I took a look at your project file and I can reproduce your issue. 
I couldn't narrow down which setting specifically impacted your crash, I'll have to bring it up to the team for analysis.

I did see that if you slightly increase the model you are able to slice it. 
I hope that helps you get unstuck. 💪 
[image]

I'll bring it up to the team to see what they think is causing hte crash. 
Fingers crossed 🤞

truegret, 2022-12-23:
Greetings MariMakes !
Thank you.  I tried scaling my model up by 5% as you did, and I am able to slice it just fine.  I had forgotten I enabled the "Fuzzy Skin" feature (I think I used it for a scarf for a snowman).  Thanks again and Merry Christmas and Happy Holidays!!

LeadMuffins, 2022-12-26:
I had this problem too, and simply moving the model to a different position than the only it loaded into Cura on allowed it to slice successfully. Failed every time before moving the model forward a few mm.

truegret, 2022-12-26:
Hello LeadMuffins -
I had tried the following with the same .3mf project file in this ticket.  My "home" is the center of my bed.  

1) I moved the model in both the X-Y 10.00mm, re-sliced, and earned the same error.
2) I closed Cura, re-loaded the .3mf as a project, moved the model in the X-Y and Z to X10.00, Y10.00, Z0.02, sliced, and was successful.
3) I closed Cura, re-loaded the .3mf as a project, moved the model to Z0.02 (left X-Y at 0.00-0.00), sliced, and was successful.
4) Finally, I closed Cura, re-loaded the .3mf as a project, rotated the model 2° CCW as you look from above the model, left X-Y-Z at 0.00-0.00-0.00), and was successful.

I hope this helps narrow down the root cause.
Cheers

MariMakes, 2022-12-30:
Hey @truegret,

I discussed your issue with the team.
We've added a ticket to the backlog with the intent to improve since it only seems to be happening with the new tree support build.  For internal reference CURA-10095

Thanks for the report! 👍

ThomasRahm, 2023-01-06:
The crash seems to happen when generating Support Infill lines. It occurs in `SkeletalTrapezodiation.cpp`in `constructFromPolygons` when the `vertex` of the `twin` of `vd_edge` is `NULL`. Not sure what this means exactly, but it occurs when generating Tree Support, because it requests infill from the Infill class (Which should be fine).

In short, i think this bug is not in the TreeSupport code. I suspect it is the same bug that causes slicing failures, that can be avoided by moving the model around a bit.

DerGenaue, 2023-02-03:
FYI: This one should probably be fixed in the next version together with the "center of buildplate" issue

truegret, 2023-02-03:
Awesome!  Yea, I did read, and tried some of the work arounds (for example moving the model on the bed), and they got me through and allowed me to print.  Thanks so much.  I really appreciate the dev team and the continued efforts to maintain a quality product!

Vandresc, 2023-04-21:
Hi all, this issue (for this specific case) is now fixed. Thank you all for your feedback!

closed 2023-04-21 by Vandresc