Ultimaker/Cura #14114: Slicing fehlgeschlagen

Frami1960, 2022-12-22:
### Application Version

5.2.1

### Platform

Windows 10

### Printer

Ender 3 v2 mit CR Touch

### Reproduction steps

1. Stl Datei geladen
2. Stl Datei slicing

### Actual results

Slicing fehlgeschlagen

### Expected results

kein slicing der stl datei möglich

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

1

MariMakes, 2022-12-23:
Hey @Frami1960,

Welcome to the Ultimaker Cura Github 🚀 
Thanks for your report 👍 

We'll need a little bit more information from you before troubleshooting.

Do you have a project file for us? It contains the printer and settings we need for troubleshooting. 
To save a project file go to File -> Save project. Please make sure to .zip your project file. For big files, you may need to use WeTransfer or similar file-sharing sites.

You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser. 

Danke 🙏

Frami1960, 2022-12-23:
[CE3E3V2_TDA0473 Beautiful Girl 07 142mm.zip](https://github.com/Ultimaker/Cura/files/10296878/CE3E3V2_TDA0473.Beautiful.Girl.07.142mm.zip)

MariMakes, 2022-12-29:
Hey @Frami1960,

Thanks for sharing your model👍
I can reproduce your error on my side with Cura 5.2.
I'll have someone from the team take a look.

I noticed that I was able to slice the model in Cura 4.13, I hope this helps you get unstuck.
[image]

I also noticed that the model is watertight but I'm getting some model errors.
So perhaps if you own the CAD file you can doublecheck your export settings.

MariMakes, 2022-12-29:
[image]

Oh! I checked again, there is something wrong with how her hair is rendered.
Could it be that it's a bunch of surfaces instead of a solid?

closed 2023-01-20 by github-actions[bot]