Ultimaker/Cura #14116: Pos processing modify G-code pause at height and change filament do not work.

unclebob64, 2022-12-23:
### Application Version

5.2.1

### Platform

The printer is Marlin the computer is Windows 10

### Printer

Longer LK 5 PRO

### Reproduction steps

Selected extensions post processing modify G code add a script. First script I tried was pause at height and all the correct settings it paused for two seconds and went right back to printing. Next I try to change filament put all the correct settings in it just kept printing both times it showed that the code was active. This happens whether I'm using the SD card or printing by USB cable.

### Actual results

Nothing on filament change it just keeps printing. On posit heights it'll pause for two seconds and start printing right again.

### Expected results

It should pause until I change filament until it to resume.

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

Sorry don't have the files to upload just wanted to know if you've ever heard of this problem and do you have a workaround.

GregValiant, 2022-12-23:
Thanks for the report.
I see you have a Longer but it may have the same firmware problem that some of the Creality printers suffered from.  Commands that send a message to the LCD (M0, M1, M600, M117) didn't work because the firmware on the LCD was configured incorrectly.  So the normal pause and filament change commands are ignored by the printer.
Pause at Height adds an M109 line after the pause and that can add a 10 second delay while the printer decides that the hot end is up to temperature.  It's not a real pause though.

One workaround was to use the "BQ M25" command in the Pause at Height dialog box.  The printer should park and stop but you might have to click on "Pause" in the LCD menu and then "Resume Print" to get the print going again.  You will need to do a test to see if that works.  M25 only works when printing from the SD card.

Another workaround is to put "G4 S300" (S is in seconds) in the "Gcode after pause" box.  That is a "fixed time" pause and the printer will restart at the end of the time period.  Hopefully the M25 command will work for you.

I use pause at Height a lot and I know the commands are being entered into the gcode and the printer is seeing the commands, they just don't work.  So this isn't likely to be a Cura bug but rather a firmware/LCD firmware issue.

I'll leave this open so you can respond here but I'm going to remove the bug label.  Did you check over on Reddit to see if any other Longer printers have this problem?

unclebob64, 2022-12-24:
"BQ M25" works great you would think that longer would have solved this problem with the update on the firmware I just did. I updated the motherboard and the LCD straight from longer and still have the same problem. If you hear of anybody who knows a way to solve this issue drop me a line and let me know other than that happy holidays and thank you.

GregValiant, 2022-12-24:
That's good that M25 works because the G4 workaround can be scary unless you happen to be at the printer when it pauses.  The other workaround I came up with involves a complicated Search And Replace post-processing script.
I'll keep this in mind but working with Longer to solve the problem is your best bet since it would involve another firmware update.
If it's the LCD firmware that is the problem then I don't know if that side of it can be updated as the problem is in the display board rather than the mainboard.
I'll go ahead and close this.  Happy Holidays.

closed 2022-12-24 by GregValiant