Ultimaker/Cura #14118: Crazy extra steps to print a single line

mutrised, 2022-12-23:
### Application Version

5.1.1

### Platform

windows 10

### Printer

Creality ender 5 plus

### Reproduction steps

Adding a model picked from internet (printed A LOT by its designer without issue).
The printer makes a lot of small lines for a single long one (almost got to vibration...), this leads to lower print speed and lower quality...
I'll make a python script to cleanup up the gcode, that's quite easy to spot (constant move/extusion ratio, on a straight line...), but I don't get why Cura is doing this extra work.

See for example (Y don't move):

x | E | dx | de | de/dx
-- | -- | -- | -- | --
246,329 | 2498,7916 |   |   |  
242,369 | 2499,1459 | -3,96 | 0,35427 | -0,08946
238,409 | 2499,5002 | -3,96 | 0,35427 | -0,08946
234,449 | 2499,8544 | -3,96 | 0,35426 | -0,08946
230,489 | 2500,2087 | -3,96 | 0,35427 | -0,08946
226,529 | 2500,563 | -3,96 | 0,35427 | -0,08946
222,569 | 2500,9172 | -3,96 | 0,35426 | -0,08946
218,609 | 2501,2715 | -3,96 | 0,35427 | -0,08946
214,649 | 2501,6258 | -3,96 | 0,35427 | -0,08946
210,689 | 2501,98 | -3,96 | 0,35427 | -0,08946
206,729 | 2502,3343 | -3,96 | 0,35426 | -0,08946
202,769 | 2502,6886 | -3,96 | 0,35427 | -0,08946
198,809 | 2503,0428 | -3,96 | 0,35427 | -0,08946
194,849 | 2503,3971 | -3,96 | 0,35427 | -0,08946
190,889 | 2503,7514 | -3,96 | 0,35426 | -0,08946
186,929 | 2504,1056 | -3,96 | 0,35427 | -0,08946
182,969 | 2504,4599 | -3,96 | 0,35427 | -0,08946
179,009 | 2504,8142 | -3,96 | 0,35427 | -0,08946
175,049 | 2505,1684 | -3,96 | 0,35426 | -0,08946

I tried a few things, this design is intended to be printed fast on 0.6mm nozzle with 0.75mm line width. 
I have similar design (smaller boxed), I don't have this issue on these.
The issues seems to be occuring randomly: I'll have it on X moves on 5 or 6 layers, then nothing, then on Y for again a few layers.
[CE5P_box_2x3.zip](https://github.com/Ultimaker/Cura/files/10294886/CE5P_box_2x3.zip)

### Actual results

Having 1 G1 command with constant speed and extrusion for every side of the box

### Expected results

extra steps added, splitting the line into 20 parts, leading to slower print and impacts print quality

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

I added my project file and the generated gcode. Maybe someone else trying this won't get the same issue so the gcode is given as reference to my problem.
I had to zip both files, otherwise I got unsupported file type error on upload...

MariMakes, 2022-12-23:
Hey @mutrised,

Thanks for your report 👍
This is outside my field of expertise. 
I wouldn't know how to help 😓 

I forwarded your bug to our Print Profile Experts. 
They might be able to see what is happening 💪

pkuiper-ultimaker, 2023-01-10:
Hey @mutrised,

For some reason the attached 3mf file does not slice for me in Cura 5.1.1. It does slice in Cura 5.2 and I can see that the model walls of the inner pockets are getting thinner when moving upwards. At layer 23 they show subtle variations in linewidth:

[image]

The slicer reacts by chopping up these lines in segments of slightly different width. However the speed changes between these segments are low and should not cause vibrations. It could be that the Ender cannot cope with the high amount of segments per second and stops the motion to catch up again (buffer underrun). You can merge the split line segments again by increasing the maximum extrusion area deviation allowed to merge lines from 50.000um2 to 5.000.000um2:

[image]

This should solve your problem.

mutrised, 2023-01-13:
Hi,

Thanks for the hint, I'll try this and post feedback here.