Ultimaker/Cura #14120: Slicing Failed with an Unexpected error.

helakejr, 2022-12-23:
### Application Version

5.2.1

### Platform

Windows 11 x64

### Printer

Biqu B1 SE Plus

### Reproduction steps

1. Load stl file (MomsCookieCutterModel5a.stl)
2. Slice
[image]

### Actual results

3. Cura displays "Slicing Failed with an unexpected error. Please consider reporting a bug on our issue tracker."
[image]

### Expected results

Slice model - producing gcode file

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[MomsCookieCutterModelv5a.zip](https://github.com/Ultimaker/Cura/files/10296004/MomsCookieCutterModelv5a.zip)

helakejr, 2022-12-23:
I found a Work around. 
1. Select model, 
2. lay model flat, 
3. and slice works as normal.
4. Note: Also verified bug and work around in Cura 5.1.0 as well as Cura 5.2.1.

MariMakes, 2023-02-21:
Hey @helakejr,

Welcome to the Ultimaker Cura Github 🚀 
Sorry, it took us a while to get back to you 😞 
I hope you still got a chance to eat your santa cookies in time for christmas 🎅 

This seems to be a duplicate of https://github.com/Ultimaker/Cura/issues/13643
This has been resolved in the upcoming Cura 5.3 Beta. 
[image]

I'll close this issue since there is a solution in place.

closed 2023-02-21 by MariMakes