Ultimaker/Cura #14128: model inserting issue

ojibs11, 2022-12-24:
### Application Version

5.2.1

### Platform

macOS monterey

### Printer

ender 3 pro

### Reproduction steps

wont insert model.
redownloaded model and still wont insert

### Actual results

file name changes to what the model is but nothing inserted on build plate
[image]

### Expected results

model should be seen

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[image]

GregValiant, 2022-12-24:
Thanks for the report.
The model might be in there but so far off screen it doesn't show.  Try this:
Right click and select "Select all Models".
On the left toolbar select the Move tool and if the numbers are big then the model is there.  Set all the numbers to "0".

If that isn't the case then something else is wrong and someone from the Cura team may be able to help.

ojibs11, 2022-12-25:
thanks this fixed the issue with this new update it made the model so small that you couldn't actually see it when it was inserted

closed 2022-12-25 by ojibs11

GregValiant, 2022-12-25:
One of the tools you get with Mesh Tools is an option for default scaling when models are brought in.  It's a really good set of tools to have.  It's in the Cura Marketplace.  After installing you will have to quit Cura and restart it.
After the restart go to "Extensions" and "Mesh Tools" and down at the bottom is Mesh Tools Settings.  One box is "Units for models that don't specify units".  If models are drawn in inches then you would select Inches and Cura will auto scale small models by 2540% to convert them to millimeters.