Ultimaker/Cura #14129: Cura 5.2.1 GUI Text shrinks after being left running all day

CZKR, 2022-12-25:
### Application Version

5.2.1

### Platform

Windows 10

### Printer

Ender 3 S1 Pro

### Reproduction steps

1. Open Cura 5.2.1 as normal (you can display a model or not
2. Switch to other Windows apps and leave Cura running in the background all day
3. Come back into Cura later - GUI interface text is about 75% of original start size
4. Workaround to fix is to close and restart Cura - resets the GUI text size
5. See attached screenshot - top shows shrunk text after running all day (captured in Windows Snipping tool at actual size), while bottom shows a freshly restarted Cura session overlapping the Snipping tool window for comparison.
[image]

### Actual results

GUI is still usable but very hard to read

### Expected results

see above

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/10299319/cura.log)

DerGenaue, 2022-12-25:
Did you possibly change the UI scale in Windows for this Screen the application is on in the meantime?

fieldOfView, 2022-12-25:
Do you have multiple displays with different DPI settings?

CZKR, 2022-12-27:
> Do you have multiple displays with different DPI settings?

Yes I do.  A main monitor running native resolution 2560x1440 but with Windows Display Scale and layout set to 125% for "Change the size of text, apps and other items", and a secondary monitor running native resolution of 1920x1080  (scale and layout set to 100%).  Cura is running on the 2560x1440 @125% monitor, and this is where I leave it running when the text shrinking occurs.  FYI:  the reason I set the higher resolution monitor to 125% is so I can better read the text as my eyesight has worsened with age, but I still get the benefit of awesome graphics (eg. gaming on a EVGA RTX 3070Ti).  Eventually I'll get a replacement 2nd monitor with 2560x1440 but I'm not ready to just yet.

CZKR, 2022-12-27:
> Do you have multiple displays with different DPI settings?

Yes - see my response to @fieldOfView for details.

CZKR, 2022-12-27:
> Yes - see my response to @fieldOfView for details.

CZKR, 2022-12-27:
@DerGenaue Yes - see my response to @fieldOfView

DerGenaue, 2022-12-27:
Usually after every change of UI Scale, you have to restart some applications to apply the new scale.
Apparently, Cura is one of them.
If it still happens after some time of not changing that setting at all, that would be weird.

CZKR, 2022-12-28:
@DerGenaue I've had the 2560x1440 monitor set to 125% display scale for many months and don't change that setting ever.