Ultimaker/Cura #14131: Temperature settings

TonyVinci0112, 2022-12-25:
### Application Version

4.9.1

### Platform

Windows 7

### Printer

Ender 5 Plus

### Reproduction steps

1. I've set my GCode flavor to Ultimaker 2
2. Material temperature settings have dissapeared from visible settings even thou I've set "check all" in my settings visibility
3. Temperature settings appear again if I change gcode flavor to another one.
[image]

### Actual results

You can see that there is a bug with temperature settings visibility on Ultimaker 2 gcode flavor profile

### Expected results

All the settings must be visible

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Hi! Sorry if something is wrong. This is my first time I report a bug in cura. Also sorry me if my english is bad cause it's not my native language. I just saw that temperature settings disappear if i set gcode flavour to Ultimaker 2. Maybe I don't know something and this is ok. Hope this would be useful if it's not ok.

GregValiant, 2022-12-25:
Thanks for the report.
I'm not an Ultimaker guy but what I believe is that for all UM printers the "Material" is passed to the printer and the printer knows the temperatures for that specific material.  As a consequence there is no need for temperature settings in Cura itself.

If that is incorrect then I'm sure I'll be suitably chastised but since it's a holiday and I don't know when the Cura team will return to work I thought I'd respond so you would kind of know what is going on.  I think what you have there is "expected behavior" as it is an UltiMaker printer.

MariMakes, 2022-12-30:
Hey @TonyVinci0112,

Welcome to the Ultimaker Cura Github 🚀 
Thanks for your report 👍

I was surprised to see a screenshot of Cura 4.9, but if you are stuck on Windows 7, the highest Cura version you can go to is indeed Cura 4.10 https://support.makerbot.com/s/article/1667337917781

@GregValiant was almost right! 😉 UltiMaker generations UM3, UM2+C, S3, and S5 can receive Material Profiles. 
For the Ultimaker 2, materials are handled on the printer, not in Cura. The idea was to be able to slice once and print with any material. Since then we have all come to the realization that it does not work like that. You need to slice differently for different materials. By selecting Ultimaker 2 as gcode flavor you are removing the material-specific settings we ended up adding later. 

The gcode flavor that you need is Marlin instead of Ultimaker 2. 
If you want to print on an Ender 5 I suggest creating a custom printer based on the current printer definitions: https://github.com/Ultimaker/Cura/blob/main/resources/definitions/creality_ender5plus.def.json and https://github.com/Ultimaker/Cura/blob/main/resources/definitions/creality_base.def.json

Can you let us know if this helps you get unstuck? 🙏

closed 2023-01-20 by github-actions[bot]