Ultimaker/Cura #14133: Xmas Alpha Tree supports taking forever to slice when Tip Dimeter is less than 0.8mm

JJPtech, 2022-12-25:
### Application Version

5.3.0 Alpha+Xmas

### Platform

Windows 10

### Printer

CR-10V3

### Reproduction steps

I recently enabled the new Thomas Rahm Tree supports with adaptive layers on by default, using a 0.4mm nozzle. I noticed that when I change the Tree support Tip Diameter from the default 0.8mm to 0.4mm, the slicing process takes an excessively long time, and even after one hour it doesn't produce any gcode.

I have attached three files to this message:

Parameters Used that work
[Parameters Used that work.zip](https://github.com/Ultimaker/Cura/files/10300639/Parameters.Used.that.work.zip)

Parameters Used that takes forever to Slice:
[Skull Bad parameters.zip](https://github.com/Ultimaker/Cura/files/10300641/Skull.Bad.parameters.zip)

Attached the files of the Project doesn't produce a gcode even after 1 hour
[Skull Not Slicing.zip](https://github.com/Ultimaker/Cura/files/10300646/Skull.Not.Slicing.zip)

### Actual results

The slicer took almost 1 hour without actually producing a Sliced Gcode

### Expected results

It should produce a Gcode

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

No aditionals, thanks

ThomasRahm, 2022-12-26:
The bug is caused by `Limit Branch Reach`. You should be able to slice if you disable it. It still is very slow for some reason i haven't figured out (took 2 minutes on my system).

JJPtech, 2022-12-26:
Yes, it woks.
I noticed that the slicing process is taking longer than usual as well as you, when using the new Thomas Rahm Tree supports. It is possible that this is due to the increased computational power needed to generate the more complex and optimized shapes offered by these tree supports.

Amazing implementation the way. Thanks

ThomasRahm, 2022-12-26:
There is something wrong with `support pattern` `zigzag` i think. Don't know why, but it generates far to many branch-tips, so the performance is bad. `Triangle` for example slices much faster. Also the support interface no longer intersects with support at the top of the skull.

ThomasRahm, 2022-12-27:
Relevant for the devs at Ultimaker: 
The issue causing the slicing to not terminate was a rounding error. See this commit for a fix: https://github.com/ThomasRahm/CuraEngine/commit/201a12dbad708730aed20a0b9093c8a219079657

JJPtech, 2022-12-27:
Glad that my silly post helped collaborated a bit. Again, this feature is a huge improvement. Thanks! 

I don't want to close this posy by my self in case someone can benefit from this. or add something.

MariMakes, 2022-12-29:
Just jumping in to say that I'm loving the interaction here. ❤️ 
I'll make sure that the Devs see your commit, @ThomasRahm 👍

MariMakes, 2022-12-30:
We've added a ticket to the backlog to merge the fix. 
For internal reference CURA-10091

Thanks for the report and the quick fix! 👍

DerGenaue, 2023-07-04:
Should be solved in 5.4

MariMakes, 2023-07-11:
True 👍
I'll close this issue.

closed 2023-07-11 by MariMakes