Ultimaker/Cura #14134: Empty slice with adaptive layers and object mesh is elevated

mirakct, 2022-12-25:
### Application Version

5.2.1

### Platform

Windows 10

### Printer

Custom

### Reproduction steps

1. Activate adaptive layers
2. Elevate part by Z axis translation
3. press slice

### Actual results

1. Slicing process finishes without errors
2. 0 Minute print time / no moves

### Expected results

G-Code will be generated / motions sequenced

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

expected results, when dropping part to buildplate or disabling adaptive layers

GregValiant, 2022-12-26:
Thanks for the report.
We need a project file to troubleshoot or we'd just be guessing.
Load the model, set Cura up to slice it.  Use the "File | Save Project" command to create a 3mf project file.  Zip the 3mf file and post the Zip folder here.

mirakct, 2022-12-27:
[bug.zip](https://github.com/Ultimaker/Cura/files/10308168/bug.zip)

Thanks for the quick workaround, that worked for me. Find attached the requested project file.

MariMakes, 2022-12-29:
Thanks for the report 👍 
We've had quite some reports on this and it was long ruled as a known issue. 
We've been able to resolve the issue earlier this week, here https://github.com/Ultimaker/CuraEngine/pull/1787 and you can expect it in the new Stable release 🎉

closed 2022-12-29 by MariMakes