Ultimaker/Cura #14138: (5.2.1) STL slices correctly when imported, but not when mirrored.  Tree supports

Austenite2, 2022-12-26:
### Application Version

5.2.1

### Platform

Windows 10

### Printer

Voron2 300

### Reproduction steps

Created an STL, from Fusion 360
Import into Cura
Sliced with tree supports, works fine
Mirrored STL, try to slice again/
Error message: Slicing failed with an unexpected error.  Please consider filing a bug report

### Actual results

Error message: Slicing failed with an unexpected error.  Please consider filing a bug report

### Expected results

Should have sliced and produced gcode.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[V300_Hat stand_mirror.zip](https://github.com/Ultimaker/Cura/files/10305013/V300_Hat.stand_mirror.zip)

Austenite2, 2022-12-26:
Object in attached project file was mirrored in Cura using the red horizontal arrows in the interface.  When mirrored using the green vertical arrows, slicing completes correctly.

MariMakes, 2023-01-05:
Hey @Austenite2,

Welcome to the Ultimaker Cura Github 🚀 
Thanks for your report 👍

I think I can help you! Something seems wrong with your normals after mirroring. 
You can fix that with the Meshtools plugin
You can get it here: https://marketplace.ultimaker.com/app/cura/plugins/fieldofview/MeshTools

[image]

After the fix I was able to slice your file succesfully.
[image]

Since you are slicing with tree support have you considered testing the new Tree Support Alpha and sharing any feedback with us? You can download a version here: https://github.com/Ultimaker/Cura/releases

To me, it doesn't seem like a bug in Ultimaker Cura, since we have a solution in place. 
Would you still like to keep this issue open or do you agree that we can close it?

closed 2023-01-20 by github-actions[bot]