Ultimaker/Cura #14140: Any news on a more consistent Z-seam?

meltiseugen, 2022-12-27:
### Application Version

5.2.1

### Platform

All

### Printer

All

### Reproduction steps

Add a sphere, slice it with z-seam set as: user specified, seam corner preference as None

added project file

[sphere.3mf.zip](https://github.com/Ultimaker/Cura/files/10306731/sphere.3mf.zip)

### Actual results

zig-zag seam positioning

[image]

### Expected results

straight seam positioning

[image]

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

N/A

MariMakes, 2022-12-30:
Hey @meltiseugen,

Thanks for your report 👍

I already spoiled it a little in this comment https://github.com/Ultimaker/Cura/issues/11877#issuecomment-1367219842 but we expect improvements in the zseam in Cura 5.3 stable.

I expect your model to look like this in the new version of Cura. 
[image]
It's not completely resolved, but we are getting closer to a solution.

Would you still like to keep this issue open or do you agree that we can close it?

meltiseugen, 2022-12-30:
> Hey @meltiseugen,
> 
> Thanks for your report 👍
> 
> I already spoiled it a little in this comment [#11877 (comment)](https://github.com/Ultimaker/Cura/issues/11877#issuecomment-1367219842) but we expect improvements in the zseam in Cura 5.3 stable.
> 
> I expect your model to look like this in the new version of Cura. [image] It's not completely resolved, but we are getting closer to a solution.
> 
> Would you still like to keep this issue open or do you agree that we can close it?

Well, you did provide news on the seam problem haha. So let's meet in the middle and I can close it. 

Honestly, I can't wait for the issue to be completely fixed (5.4?). But I really like the iterations :)

Thank you!

closed 2022-12-30 by meltiseugen