Ultimaker/Cura #14142: Weird behavior with adaptive layers enabled

fabianoriccardi, 2022-12-27:
### Application Version

5.2.1

### Platform

Windows 10

### Printer

Creality Ender 3

### Reproduction steps

1. Import the attached model
2. Enabled Adaptive Layers
3. Don't touch any other options

This model is a simple cube caved with a cylinder. The base of the cube is 0,4mm. 
[Body1.zip](https://github.com/Ultimaker/Cura/files/10307493/Body1.zip)

### Actual results

The "base" should be 0.4mm (base layer height 0,2mm => 2 layers), but actually there is only one layer.

### Expected results

The base of the cube should be composed by 2 layers. If *adaptive layers* is disabled, the base is composed, as expected, by 2 layers.

[image]

[Body1.zip](https://github.com/Ultimaker/Cura/files/10307498/Body1.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

No other info.

GregValiant, 2022-12-28:
Thanks for the report.
Given your settings this is expected behavior.  You have the "Adaptive Layers Maximum Variation" set to 0.2mm.  Since there are no topography changes in the model that require the layer height to be made smaller, and since your layer height is 0.2mm, Cura is calculating the layers upward in .04 increments (your "Variation Step Size")  until it hits the 0.4 layer height (your regular layer height of 0.2 + the maximum variation of 0.2 = 0.4mm).
You will also notice that with Adaptive turned off there are 50 layers, but with adaptive turned on there are only 26 layers.  Cura is adapting the height of a couple of the bottom layers upward to hit the max layer height you have set to 0.4.  Since Layer 2 goes down at .24 layer height it misses that roof of the bottom of the hole and leaves it with a single layer at 0.2 high.

I'll close this as it is a settings issue rather than a Cura bug.

closed 2022-12-28 by GregValiant

fabianoriccardi, 2022-12-28:
Sorry Greg, I was making different experiments and I uploaded the not-clean 3mf version, but the result is the same if I use the default options for *Adaptive Layers Max Variation*, *Adaptive Layers Variation Step Size* and *Adaptive Layers Topography Size*. Here the clean version of 3mf file:

[Body1-updated.zip](https://github.com/Ultimaker/Cura/files/10315670/Body1-updated.zip)

However, thanks for the explanation, but I don't understand why this happens:

> Since Layer 2 goes down at .24 layer height it misses that roof of the bottom of the hole and leaves it with a single layer at 0.2 high.

This shouldn't happen since topography is changing!