Ultimaker/Cura #14144: Failing to discretize parabola! Must add an apex or one of the endpoints.

Block141, 2022-12-27:
### Application Version

5.2.1

### Platform

MacOS w/ AMD Radeon Pro 5500M 4GB

### Printer

Ender 3 Pro

### Reproduction steps

1. Imported stl.
2. Attempted to slice.

### Actual results

Error message, "slicing failed with unexpected error. Please consider reporting a bug." The log file states, "Failing to discretize parabola! Must add an apex or one of the endpoints." Then, killing engine.

### Expected results

It normally slices fine. Just while I'm frustrated, I thought 3D printing was supposed to be fun, but it has been more software and firmware trouble shooting than it has been printing. This all has been kind of a bummer. Rant done. I feel a bit better, thanks for letting me do that.
[bugreport.zip](https://github.com/Ultimaker/Cura/files/10309890/bugreport.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

I'm pretty new to this so if it was user error of me modifying something in the .stl blender then I apologize in advance. Any advice then would be greatly appreciated (trying to learn all aspects of printing).
[bugreport.zip](https://github.com/Ultimaker/Cura/files/10309950/bugreport.zip)

MariMakes, 2023-01-05:
Hey @Block141,

Welcome to the Ultimaker Cura Github 🚀 
Sorry that you've been experiencing so much trouble. 
I'm here to help 💪 

I'm getting an error that your mesh is faulty. 
Have you tried doing a quick mesh fix with the Meshtools Plugin? 
You can get it here: https://marketplace.ultimaker.com/app/cura/plugins/fieldofview/MeshTools
If that doesn't fix it you can try a more elaborate mesh fixer like https://myminifactory.github.io/Fast-Quadric-Mesh-Simplification/.

Also, I haven't been able to reproduce your crash with my default printer settings. 
Do you have a project file for us? It contains the printer and settings we need for troubleshooting. 
To save a project file go to File -> Save project. Please make sure to .zip your project file. For big files, you may need to use WeTransfer or similar file-sharing sites.

You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser.

closed 2023-01-06 by Block141

Serneum, 2023-01-25:
Commenting here per the Bug Report form since I am running into a similar issue.

### Application Version
I have run into this on Cura 5.2.1 and 5.2.2

### Platform
Windows 11

### Printer
Creality Ender 3

### Reproduction steps
1. Download huts-furs-85.stl from [Printables.com](https://www.printables.com/model/62856-hate-board-game-terrain-huts-plateaus/)
2. Load STL into Cura
3. Choose Super Quality or Standard Quality profile
4. Slice the file

### Actual results
Cura reports an error during slicing:
[image]

### Expected results
The file should slice. I have been slicing and printing plenty of files without issue

### Addition information & file uploads
In the logs, I am seeing a `Failing to discretize parabola! Must add an apex or one of the endpoints.` message consistently when there are slicing failures.

I am using the profiles from [Fat Dragon Games](https://www.drivethrurpg.com/product/372964/FDG-Custom-Cura-Profile-for-Creality-Printers), but I am unsure if that has anything to do with the issue. I can successfully slice the file with the default Dynamic Quality and Low Quality profiles, but not the default Standard Quality or Super Quality profiles.

I also tried simplifying the mesh after coming across [a similar issue](https://github.com/Ultimaker/Cura/issues/14144) but that file did not work either.

I have included the profile I am trying to use in the .zip file, along with the project file and logs
[Cura-debugging.zip](https://github.com/Ultimaker/Cura/files/10496628/Cura-debugging.zip)