Ultimaker/Cura #14145: No "File Open"-Dialog on 5.0.0 and up including 5.3.0-alpha+xmas

zwieblum, 2022-12-27:
### Application Version

5.2.1

### Platform

Linux

### Printer

any

### Reproduction steps

All versions of Cura beginning with 5.0.0 and including 5.3.0-alpha don't shoe a "File Open"-Dialog on Debian and Devuan Linux. This is most likely a bug in QT6.

### Actual results

Press "file open"-Icon. No "file open"-dialog appears.

### Expected results

A "File Open" Dialog should appear.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Need Debian stable :)

MariMakes, 2022-12-30:
Hey @zwieblum,

Welcome to the Ultimaker Cura Github 🚀 
Thanks for your report 👍

This reminds me of https://github.com/Ultimaker/Cura/issues/12659 where other people have reported that they've had trouble with saving files on Linux. 

I'll bring it up to the team to see if it's related. 

Can you check https://github.com/Ultimaker/Cura/issues/12659 to see if there are workarounds there to help you get unstuck? Thank you 😊

zwieblum, 2022-12-30:
As the "File Save"-dialog is the same type of dialog as the "File Open"- dialog I'd say the cause is the same. When I run the code example from https://github.com/Ultimaker/Cura/issues/12659#issuecomment-1173129911 then I do not get any dialog. However, d.exec() returns 0.

Swivelgames, 2022-12-30:
I have this same issue. Clicking `Open File(s)...` does absolutely nothing. I don't get any errors in the logs either.

MariMakes, 2023-01-10:
Hey all,

Do you still have the option to drag and drop your model into Cura or are you completely blocked? 

I've brought it up to the team and we believe it's the same issue as mentioned here https://github.com/Ultimaker/Cura/issues/12659. We've received quite a number of reports so I hope we can resolve it soon.

I'll mark this as a duplicate but I'll keep it open so we can an extra eye on this flow.

homeworkprod, 2023-02-13:
I'd like to add the perspective that *the very first thing* I do after Cura has fully started is to hit Ctrl+o to load a model and that it's pretty serious that the file selector dialog does not open, rendering the whole application essentially useless for this core use case. (Unfortunately I had not tried drag'n'drop and already removed the 5.x releases from my system so I can't say if that's working, sorry.)

Add to that the time invested for having to try a number of older releases to finally find one that works (I was updating from v4.8.0), including the "modern" and non-"modern" Linux variations (which do not hint at me which one I should choose) each.

I'm on Debian as well. Can someone share if this happens on Ubuntu, too?

zwieblum, 2023-02-14:
The solution is quite easy and annoying at the same time: AppImages are not  fully self contained any more. In this case you need to install "xdg-desktop-portal-gtk" or "xdg-desktop-portal-kde" to get the dialog working again. Say thanks to GNOMES of freedesktop.org :[

homeworkprod, 2023-03-06:
> In this case you need to install "xdg-desktop-portal-gtk" or "xdg-desktop-portal-kde" to get the dialog working again.

This suggestion did not work for me, unfortunately.

Swivelgames, 2023-03-13:
Installing those didn't fix it for me either. Additionally, dragging and dropping isn't an option for everyone, so even that workaround is not suitable for me, unfortunately :slightly_frowning_face:

Looking forward to this being fixed!

MariMakes, 2023-03-31:
Quick update on our side. 

We might have a fix for this issue. 🎉 
It's a bit hard for us to test, but we have a nightly build available. 
Can you please try it out and let us know if it resolves your issue? 

[You can find the download links and the discussion here. ](https://github.com/Ultimaker/Cura/discussions/15098)

Swivelgames, 2023-04-19:
Great to hear! I'll try and pull a nightly build down tomorrow and see if it's fixed!

fermino, 2023-09-04:
Hi! I think I am experiencing the same bug here. Installing xdg-desktop-portal does solve the issue (at least for me, I'm running sway on an arch linux box, so that's wlroots), but it is kind of annoying as it requires pipewire and I'm running pulseaudio.

I tried the nightly build @Swivelgames mentioned with no luck :(

I do think it is related to QT6 as this is the stack trace. I'm willing to help debug it if necessary :)
```
Stack trace of thread 48721:
#0  0x00007f231008e83c n/a (libc.so.6 + 0x8e83c)
#1  0x00007f231003e668 raise (libc.so.6 + 0x3e668)
#2  0x00007f231003e710 n/a (libc.so.6 + 0x3e710)
#3  0x00007f21c4a7b4b2 n/a (/tmp/.mount_curaJNlEhS/libQt6QuickDialogs2QuickImpl.so.6 + 0x4e4b2)
ELF object binary architecture: AMD x86-64
```

MariMakes, 2023-10-19:
👋 Quick update on our side. 

This is resolved in the 5.5 Beta release 🎉 , you can download the Cura version with the fix here: [https://github.com/Ultimaker/Cura/releases/tag/5.5.0-beta.1](https://github.com/Ultimaker/Cura/releases/tag/5.5.0-beta.1)
I'll close this issue since it's resolved. 

Thanks again, and please let us know if you run into any other issues 💪

closed 2023-10-19 by MariMakes

MariMakes, 2023-10-20:
Oh! We are aware of an issue with the file open icon. 
We are collecting reports and tips here instead: https://github.com/Ultimaker/Cura/issues/16932

There is even a build with a potential fix there, but it's a tricky one, so feedback is welcome