Ultimaker/Cura #14146: marketplace error

sgtwebbpd, 2022-12-28:
### Application Version

5.2.1

### Platform

windows 10

### Printer

Ender 3 S1

### Reproduction steps

I clicked on the marketplace button on the top right of the screen. 

### Actual results

A second small window opened but never loaded anything within it. After a minute I tried to close the second window, but the error box popped up saying the program had stopped responding. I tired the process 4 times with the same results. I uninstalled the program and reinstalled it and the results are the same. 

### Expected results

The marketplace window would load and I could look for add-on programs. 

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

n/a

MariMakes, 2022-12-29:
Hey @sgtwebbpd,

Welcome to the Ultimaker Cura Github 🚀 
Thanks for your report 👍

We have a number of similar reports that all have slightly different causes. 
Can you share your log files so we can narrow them down? 

You can find your log file here:
Windows: %APPDATA%\cura\<Cura version>\cura.log or usually C:\Users\\<your username>\AppData\Roaming\cura\<Cura version>\cura.log

You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser.

sgtwebbpd, 2022-12-29:
> C:\Users\\AppData\Roaming\cura<Cura version>\cura.log

Hi,

I am new to this, and can't find the log files you are asking about. I
installed Cura on my D drive, but looked there and under users on my C
drive and still no luck.

Peter