Ultimaker/Cura #14150: Layer Height influences amount of Support Trees

Firyfly, 2022-12-28:
### Application Version

5.3.0-alpha-xmas

### Platform

Windows 10

### Printer

Artillery Genius Pro

### Reproduction steps

1. Imported Model 
2. Layer Height was 0.3 and Tree support was tested and sliced. Multiple Tree supports were created
3. Changed Layer Height to 0.12 and sliced, only 1 Tree support was created

### Actual results

Different amount of tree supports depending on layer height

### Expected results

Same amount of support structures even when increasing the detail of the print through the layer height. 

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[LayerHeightAffectsTreeSupportAmountIssue.zip](https://github.com/Ultimaker/Cura/files/10315235/LayerHeightAffectsTreeSupportAmountIssue.zip)

ThomasRahm, 2022-12-30:
Interesting. Some difference between different layer heights is expected. But that one model has no support at all is not.
Switching `Support Distance Priority` to `xy overrides z` causes trees to generate again.
Looking into this, i found the issue: `minimum support area`. If the cura is told to only support overhang areas larger than `5 mm^2` (on one layer) then overhangs smaller than this will not be supported.
Reason why you had support with a larger layer height is because then more of the overhang area in on one layer => Larger overhang area per layer.
I recommend setting `minimum support area` to 0 when using tree supports. Maybe there should a warning when this setting is larger than 0 and tree support is used.

Firyfly, 2022-12-30:
A big big thank you!
I tested it out and can confirm that this was the reason and upon setting minimum support area to 0 resolved the porblem and trees were created normally. 

Thanks for the fast reply and the amazing work. 
I wish you "Einen guten Rutsch", which is german for a happy new year. 
~Greetings from germany

Firyfly, 2023-01-05:
I was able to get things working, so from my side the issue is solved. 
And i definitely agree with ThomasRahm, that a little warning would be very helpful as its otherwise hard to figure out why all of a sudden nothing is being generated anymore haha

Thanks for all your work and effort!