Ultimaker/Cura #14153: 5.2.1 (or other 5. Version) didn't on MacOS 12.6.2

msman0, 2022-12-28:
### Application Version

5.2.1

### Platform

MacOS 12.6.2

### Printer

Ender 3 V2

### Reproduction steps

Install Version 5.2.1 uninstall and install again.

### Actual results

The window with the Version number and loading is displays but the then it didn't start. No error message displayed.
Version 4.13.1 works normal.

### Expected results

Program start.

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

 
[cura.log](https://github.com/Ultimaker/Cura/files/10315870/cura.log)

MariMakes, 2022-12-29:
Hey @msman0,

Welcome to the Ultimaker Cura Github 🚀 
I'm sorry to hear you are stuck 😞 

I have a couple of questions to narrow it down. Sorry if they do not apply to your operating system. 
	1. Do you have multiple monitors? Does Cura start if you disable one of the monitors?
	2. Do you have GPU Tweak OSD overlay enabled? Does Cura start if you disable that software?
	3. Do you have a Ryzen Graphics card? Can you try this workaround? https://support.makerbot.com/s/article/1667412437627
	4. Do you have the Nahimic audio enhancement service enabled? Can you try this workaround? https://github.com/Ultimaker/Cura/issues/12102#issuecomment-1157907665

Let me know if it doesn't work. 
Than I'll have someone look at your logs.

msman0, 2022-12-30:
Hello @MariMakes,
thank you. I will try to answer:
1. yes I have a MacBookPro and installed an external Display. I have just tried it "standalone" without any other connections but the result is the same. It didn't start
2. Where can I enable/disable GPU Teach OSD overlay?
3. no only the standard MacBook Pro with internal graphic
4. I don't know, if I understand it the Nahimic audio is only in Windows systems?

Thank you and best regards.

NicNull, 2023-11-03:
I have the same issue on a Macbook Pro (2012  Mojave 10.14.6, Nvidia GPU, No external monitor). 
Works on a newer Macbook Pro (2019 Ventura 13.5, AMD radeon GPU)