Ultimaker/Cura #14155: Part wont slice

mattatouille717, 2022-12-28:
### Application Version

5.1.2

### Platform

windows 10

### Printer

ultimaker s5

### Reproduction steps

I tried to slice a model that I have sliced and printed many times before. Now it seems to not want to slice properly.

### Actual results

File wont slice.

### Expected results

file shouldve sliced

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

none

MariMakes, 2022-12-29:
Hey @mattatouille717,

Welcome to the Ultimaker Cura Github 🚀 
Thanks for your report 👍

Do you have a project file for us? It contains the printer and settings we need for troubleshooting. 
To save a project file go to File -> Save project. Please make sure to .zip your project file. For big files, you may need to use WeTransfer or similar file-sharing sites.

You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser. 

Thank you 🙏

mattatouille717, 2022-12-29:
[UMS5_1 x 1 Mounting Board pt 1 (project file).zip](https://github.com/Ultimaker/Cura/files/10320392/UMS5_1.x.1.Mounting.Board.pt.1.project.file.zip)

MariMakes, 2022-12-30:
Hey @mattatouille717,

Thanks for the update!
Wow that's an extreme printjob. 

I saw that you had tree support enabled for your PVA, so I changed it to normal support. 
It took 25 minutes to slice and ended up with this printjob: 
[image]
[Mari_UMS5_1 x 1 Mounting Board pt 1 (ufp).zip](https://github.com/Ultimaker/Cura/files/10325210/Mari_UMS5_1.x.1.Mounting.Board.pt.1.ufp.zip)

You can try to print it, but the dark blue lines indicate unretracted moves that will result in stringing. 
Is there a way that you can only print part of the printjob optimizing the different settings untill you can make sure that there is no stringing? I suspect you'll have to change a number of settings related to combing and variable linewidth. You can read more about them here: https://support.ultimaker.com/hc/en-us/articles/4792077687068

This seems to be a duplicate of https://github.com/Ultimaker/Cura/issues/14108
We have a ticket on our backlog with the intent to improve this behavior.
Until it's resolved, the best workaround I can offer is slicing the printjob in Cura 5.0 or lower. 😖 

I'll be closing this issue as a duplicate but you can follow the progress here: https://github.com/Ultimaker/Cura/issues/14108

closed 2022-12-30 by MariMakes