Ultimaker/Cura #14158: Arachne engine Fails to slice intermittently

peaceleslie, 2022-12-29:
### Application Version

Ultimaker Cura Arachne engine beta 2 99.10.0

### Platform

Windows 10 64bit

### Printer

Ender 2

### Reproduction steps

I was setting up a model I had printed before and had modified.
I rotated and laid flat the model and sliced.

### Actual results

Slicing failed with an unexpected error.
reloading did the same.

Closing Cura and reloading the model worked. 

Failed on next try.

### Expected results

Slice to complete and save to disk.

[CE2_R Mirror Stepped.zip](https://github.com/Ultimaker/Cura/files/10317428/CE2_R.Mirror.Stepped.zip)
[image]
[image]
[cura.log](https://github.com/Ultimaker/Cura/files/10317483/cura.log)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

I have had no issues with Arachne.
I still have 4.13.1 installed but have not had problems before.
Usually it tells me when there are errors so I can fix them before slicing.
I am still learning how to use all this new software and I often make mistakes that take forever to work out.

The problem I am having now is that this is intermittent and tells me nothing when it fails.
A restart of Cura allowed me to slice the model but when I tried again it failed.

GregValiant, 2022-12-29:
Thanks for the report.
The Arachne Beta was rolled into Cura 5.0.0 and the latest Cura stable release is 5.2.1 (there is a 5.3.0 Xmas beta).
I didn't have any trouble slicing your model in the Windows version of 5.2.1.  I would suggest that you upgrade and let us know if that works for you.

peaceleslie, 2022-12-29:
It works great.
Takes a lot longer to print but that will be settings I guess

closed 2022-12-29 by peaceleslie