Ultimaker/Cura #14159: Slicing failed with unexpected error

cbeaumont90, 2022-12-29:
### Application Version

5.3.0 Alpha

### Platform

Mac Ventura M1

### Printer

Flsun SR

### Reproduction steps

Imported STL from reputable source, set parameters, sliced = error

### Actual results

Error and no sliced stl for printing. I am unable to upload the project file as it is too large of a file even when compressed. 

### Expected results

Sliced STL with Gcode ready for printing. 

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Unable to upload project file as it is too large even when compressed.

MariMakes, 2022-12-29:
Hey @cbeaumont90,

Welcome to the Ultimaker Cura Github 🚀 
Thanks for your report 👍

If it's so large, are you sure it's made for 3D printing? Could you have a faulty mesh? Can you try turning the model with 90 degrees and moving it away from the center of the buildplate as mentioned in https://github.com/Ultimaker/Cura/issues/13643

We would really need your project file to start helping you with troubleshooting. It contains the printer and settings we need for troubleshooting. 
To save a project file go to File -> Save project. Please make sure to .zip your project file. For big files, you may need to use WeTransfer or similar file-sharing sites.

You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser.

cbeaumont90, 2022-12-29:
[FSR_main (Cut 1).3mf.zip](https://github.com/Ultimaker/Cura/files/10321197/FSR_main.Cut.1.3mf.zip)

cbeaumont90, 2022-12-29:
I was now able to attach it.

MariMakes, 2023-01-05:
Hey @cbeaumont90,

Thanks for sharing the file. 
But the file seems to be corrupt 😕 
It seems to be incomplete. 

[image]

Can you try again to upload?

Select File 
Select Save project.
Make sure you save it as a .3mf (curaproject)
Rightclick your file and Compress to ZIP

You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser.

closed 2023-01-20 by github-actions[bot]