Ultimaker/Cura #14160: MacBook Pro M1 -- Cura does not see printer. Printer connected.

not-marco, 2022-12-29:
### Application Version

5.2.1

### Platform

Mac OS Ventura 13.1

### Printer

Anet A8

### Reproduction steps

I have 2 MacBook Pro's.
1) 2012 MacBook Pro Intel i7 -- Working fine with Ultimaker Cura 4.1.0
2) Just bought 2021 MacBook Pro 16" with M1 Pro. Cannot see printer in Cura 5.2.1. Other slicers do see it.

1) Downloaded Ultimaker Cura from website.
2) Opened Cura.
3) Set up printer from setup wizard.
4) Clicked on the "Monitor Tab"
5) Get message " In order to monitor your print from Cura, please connect the printer."

These are the same steps I have taken to set the printer up in previous versions.

### Actual results

From Cura.log:

2022-12-28 22:43:38,589 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [493]: Loaded plugin USBPrinting 1.0.2

Printer visible/responding in Arduino IDE. Printer visible/responding in Pronterface. Printer visible/responding in BuildBee. Printer visible/responding in Slic3r... and every other slicing application I can find.

Cannot see the printer in Cura 5.2.1 no matter what. Steps taken:

- Disable the "USBPrint" plugin from the Marketplace menu cog > restart Cura and Printer > enable plugin > restart Cura and printers > issue persists
- Modifying "AutoDetectBaudJob.py" with readline() rather than read_until() as it was in previous versions where I could connect > issue persists
- Replacing whole "USBPrinting" plugin folder with older version from 4.1 which works on my other Mac > issue persists
- Retrograding built in Apple USB drivers via the .kext file to a previous version from another MacOS.  > issue persists
- Updating firmware on my printer to Marlin 1.1.9. (perhaps 2.0 would fix issue, but doesn't explain why Pronterface et. all works though) > persists
- Make new user > install Cura 5.2.1 > test > issue persists
- Reinstall MacOS > new user > no apps installed except Cura 5.2.1 > issue persists

### Expected results

Use my printer as normal.

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/10318460/cura.log)

JoyceKimberly, 2022-12-30:
Same problem on my MacBook Air M1. 
Pronterface however, connects and can print/send commands just fine, so it’s not impossible on M1’s.

MariMakes, 2023-01-05:
Hey Joyce and Mark,

I recognize your report. 
This seems to be a duplicate of https://github.com/Ultimaker/Cura/issues/13732
There is a workaround available to see your USB connected printer again.
[You can find the workaround here](https://community.ultimaker.com/topic/42417-fix-for-connecting-creality-ender-3-neo-to-mac-via-usb/) 

[We are working on a way to ship the fix with the next Cura version. ](https://github.com/Ultimaker/Cura/pull/14202) 

I'll be closing this issue as a duplicate but you can follow the progress here: https://github.com/Ultimaker/Cura/issues/13732

closed 2023-01-05 by MariMakes