Ultimaker/Cura #14162: Cura 5.2.1: traveling less optimized than Cura 4.13

tguillaume02, 2022-12-29:
### Application Version

5.2.1

### Platform

Windows 10

### Printer

Geeetech A20M

### Reproduction steps

Slice attached file with Cura 4.13 and next with Cura 5.12always with attached profile.
You can see less travel with old Cura version, so the result is better.
[repro_step.zip](https://github.com/Ultimaker/Cura/files/10319829/repro_step.zip)

### Actual results

a lot of unnecessary traveling compared to the previous version of Cura ( 4.13)

### Expected results

traveling identical to Cura 4.13

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[image]

[repro_step.zip](https://github.com/Ultimaker/Cura/files/10319829/repro_step.zip)

MariMakes, 2023-01-05:
Hey @tguillaume02,

Welcome to the Ultimaker Cura Github 🚀 

Thanks for your report 👍
That's a great example, of something that has been reported internally and we plan to improve. 

Do you have a project file for us so I can bring it up tot the team? 
It contains the printer and settings we need for troubleshooting. 
To save a project file go to File -> Save project. Please make sure to .zip your project file. For big files, you may need to use WeTransfer or similar file-sharing sites.

You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser.

tguillaume02, 2023-01-20:
I have join repor_step zip file to the first message

MariMakes, 2023-01-20:
Hey @tguillaume02,

We usually ask for a project file. It contains the printer and settings we need for troubleshooting. 
But I've been able to troubleshoot your problem with the informatio you've given me. 

I believe you are seeing more travels because  the Variable Linewidth Engine in Cura 5.x makes a lot of switches between a single and a double wall. 
​​​​​​​There are settings that you can change to improve this behavior.
You can read more about them here: https://support.ultimaker.com/hc/en-us/articles/4792077687068

[image]
I've also noticed that your Retract Minimum travel is quite high and possibly causes stringing.

I created a new project file, does this solve your problem? 
[Mari_GA_Logo a 4 voix.zip](https://github.com/Ultimaker/Cura/files/10464974/Mari_GA_Logo.a.4.voix.zip)

tguillaume02, 2023-01-20:
I will try it soon.
I imported in Cura 5.2 to see the rendering, I still find more displacement than in version 4.13
This may be due to the variable line width engine but I find it a regression.
Thank you for taking the time to study my request.

MariMakes, 2023-05-04:
Hey @tguillaume02,

Did you have a chance to look at the project file provided?

To me, it doesn't seem like a bug in Ultimaker Cura anymore since there is a solution in place.
Would you still like to keep this issue open or do you agree that we can close it?

closed 2023-05-05 by tguillaume02