Ultimaker/Cura #14163: "Slicing failed with an unexpected error" but I printed this before without issues

ultrasuperman1001, 2022-12-29:
### Application Version

5.2.1

### Platform

windows 11

### Printer

Creality CR10 Max

### Reproduction steps

1) load model and change angle to fit on built plate
2) add a support 
3) resize support and move to top and bottom
4) set cutting infill to 10%
5) slice and error

This is the right side to a large lithopane banner I'm making, I printed the left side just fine so I don't know why this side is failing

### Actual results

https://drive.google.com/file/d/1ycGyLINu56yKMm0sO4OOof9mMFgTi6-d/view?usp=sharing
[cura.log](https://github.com/Ultimaker/Cura/files/10320279/cura.log)

### Expected results

a normal slice

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

attached above

ultrasuperman1001, 2022-12-29:
Update: Flipping the model 180 degrees in both directions allows it to print. Not sure why it only prints that way though?

MariMakes, 2023-01-05:
Hey @ultrasuperman1001 

Thanks for your report 👍

What a beautifully intricate model 🌟 
[image]

This seems to be a duplicate of https://github.com/Ultimaker/Cura/issues/13643
We have a ticket on our backlog with the intent to improve this behavior.
I've been told that it's related to rounding issue, by rotating the model Cura might round something up which it rounded down in the other orientation. 

I'll be closing this issue as a duplicate but you can follow the progress here: https://github.com/Ultimaker/Cura/issues/13643

closed 2023-01-05 by MariMakes