Ultimaker/Cura #14165: Issues with importing profile

jsrfo, 2022-12-29:
### Application Version

5.2.1

### Platform

Windows 10

### Printer

Sovol SV06

### Reproduction steps

1. Attempted to import fat dragon profile for the Sovol SV06
2. Got error stating "Quality type 'Super' is not compatible with current active machine definition 'fdmprinter'."
3. Attempted to import fat dragon profile for the Sovol SV01 printer.
4. Fat Dragon profile was successfully imported for the Sovol SV01.
5. Repeated steps for importing fat dragon profile for the Sovol SV06
6. Got the same error. Attempted this a couple times with the same result.

[image]

### Actual results

1. Got the above error for the SV06
2. Successfully imported the same profile for the SV01
This was using Cura V5.2.1

### Expected results

I expected the profile to be added for the SV06 printer. 

I found it very odd that the profile imported for the SV01 which is almost three years old, but not the SV06, which was just released.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

No additional comments
[image]

MariMakes, 2023-01-05:
Hey @jsrfo,

Welcome to the Ultimaker Cura Github 🚀 
[I recognize you from your Community Report! ](https://community.ultimaker.com/topic/42413-problems-importing-profiles-in-cura-521/#comment-317013) 

I can see that the Sovol SV06 printers only have been added to [the Cura Main branch early December.](https://github.com/Ultimaker/Cura/commits/main/resources/definitions/sovol_sv06.def.json)
That means that they have not been in a Cura release yet. 

I think the reason that you are seeing this error is related to how you added the Sovol SV06 to your Cura version. 
Can you share your configuration folder so I can see the way that you've added the printer? 

You can share your configuration folder by:
	1. Go to Help at the top bar of Cura
	2. Select Show Configuration Folder
	3. Select all the folders in there and create a zip OR go up a folder in your folder structure and zip your folder with the correct Cura version
You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser.

jsrfo, 2023-01-06:
Hope this is what you need. I still cannot add the profile to my SV06 printer in Cura.

[Cura 5.2 Configuration.zip](https://github.com/Ultimaker/Cura/files/10356804/Cura.5.2.Configuration.zip)

jsrfo, 2023-01-06:
Posted my files. Hope they help.

⁣Get BlueMail for Android ​

MariMakes, 2023-02-15:
Hey, quick update. 

We believe that it's not working because the profiles are not made with your printer type. 
We want to start working on sharing profiles between printer types, it's not really a bug more of a feature request. 

I've added your problem as an test case, for internal reference CURA-10261